offspot/kiwix-hotspot #132: Upload custom CSS

rgaudin, 2020-01-16:
In GitLab by @Kelson42 on Oct 17, 2017, 11:07

It should be possible to overwrite the default CSS by a custom CSS and so one allow rebranding of  the Ideascube which is currently only branded for BSF.

To do that a filepicker should be provided to be able to upload that custom CSS.

In attachment an example of custom CSS.[style-pibox.css](/uploads/caa67444c9c10e666a5788c6939a6d7d/style-pibox.css)

rgaudin, 2020-01-16:
In GitLab by @thiolliere on Nov 19, 2017, 20:41

where should we put this file ? @mgautierfr

rgaudin, 2020-01-16:
In GitLab by @mgautierfr on Nov 27, 2017, 16:31

mentioned in commit ideascube@856b256ba87c8d7bce2e22f46c313c5f178b3e07

rgaudin, 2020-01-16:
In GitLab by @mgautierfr on Nov 27, 2017, 16:31

mentioned in merge request ideascube!894

rgaudin, 2020-01-16:
In GitLab by @mgautierfr on Nov 27, 2017, 16:36

I was pretty sure that we were able to do a branding of the css but it is not the case :/

I've just create the MR ideascube/ideascube!894 to do that. Once it will be merged, putting the `style.css` file in the `branding` directory along side of `favicon.png` and `header-logo.png` (if the MR is merge without change).

However, it will request us to use the last next of ideascube :(

rgaudin, 2020-01-16:
In GitLab by @Kelson42 on Nov 27, 2017, 17:41

@mgautierfr Nice but having to use the last version of Ideascube seems to be a serious problem as this needs a lot of work. @thiolliere you confirm?

rgaudin, 2020-01-16:
In GitLab by @thiolliere on Nov 27, 2017, 18:48

no, ideascube was pretty stable and I looked quickly into the changelog, we probably just have to update ideascube version in our branch of ansiblecube.

It would have been more difficult to update to last version of ansiblecube as I would have had to rebase my modification and maybe add and remove some.

rgaudin, 2020-01-16:
In GitLab by @Kelson42 on Nov 27, 2017, 18:59

@thiolliere As providing a way to custom the CSS is almost (now we also have this pb with damm slow Windows) the last blocker for 1.0, If you can do that upgrade, that would be really great.

rgaudin, 2020-01-16:
In GitLab by @thiolliere on Dec 12, 2017, 17:15

@mgautierfr I don't have rights to modify https://framagit.org/ideascube/ansiblecube.git

what I want to change is roles/ideascube/tasks/ideascube.yml line 35:
```
-  apt: name=ideascube=0.29.2 force=yes update_cache=yes dpkg_options="force-confnew"
+  apt: name=ideascube=0.36.0 force=yes update_cache=yes dpkg_options="force-confnew"
```

rgaudin, 2020-01-16:
In GitLab by @mgautierfr on Dec 13, 2017, 10:20

@thiolliere I add you as developper to ideascube/ansiblecube.

You should have the rigth to push on the repository now.

rgaudin, 2020-01-16:
In GitLab by @thiolliere on Dec 17, 2017, 13:19

mentioned in merge request !26

rgaudin, 2020-01-16:
In GitLab by @thiolliere on Dec 17, 2017, 17:08

closed via commit 98d8774c7e25ba41802ac467b5bc31b30fabfccd

rgaudin, 2020-01-16:
In GitLab by @thiolliere on Dec 21, 2017, 22:05

mentioned in commit 98d8774c7e25ba41802ac467b5bc31b30fabfccd

closed 2020-01-16 by rgaudin

brandonpsmith, 2024-02-21:
@rgaudin Is there an example of the css styling? The file uploaded in this issue is gone.

rgaudin, 2024-02-21:
@brandonpsmith, no. If you're using it, I'd advise you work from the DOM that is generated.