offspot/kiwix-hotspot #170: Use standard OPDS catalog (instead of custom YML file)

rgaudin, 2020-01-16:
In GitLab by @Kelson42 on Apr 2, 2018, 06:46

We need an alternative to the YAML library of BSF because:
* All of their content are not free
* The catalog is not kep up2date

So, it should be possible - as a replacement - to specify a Kiwix library as input replacement.

rgaudin, 2020-01-16:
In GitLab by @mgautierfr on Apr 3, 2018, 08:09

The YAML format is the format of ideascube, not BSF. BSF have just created catolog files compatible with ideascube.

rgaudin, 2020-01-16:
In GitLab by @rgaudin on Apr 30, 2018, 17:41

mentioned in commit 8b128fa4b0bcb9475f38f1a339591bb84ec3139c

rgaudin, 2020-01-16:
In GitLab by @Kelson42 on Dec 28, 2018, 12:58

changed title from **{-Pibox-installer should be able to ingest Kiwix library (http://library.kiwix.org) catalog in OPDS format-}** to **{+Use standart OPDS catalog (instead of custom YML file)+}**

rgaudin, 2020-01-16:
In GitLab by @Kelson42 on Dec 28, 2018, 12:58

changed title from **Use standar{-t-} OPDS catalog (instead of custom YML file)** to **Use standar{+d+} OPDS catalog (instead of custom YML file)**

rgaudin, 2020-01-16:
In GitLab by @rgaudin on May 9, 2019, 10:19

Now that we have a version without ideascube, there is no blocker to implement this.

Reason would be simplification of the source library management. As matthieu pointed, the reasons mentioned above are not relevant anymore (and we drop BSF catalogs a while ago).

We would still have to maintain the `ideascube.yml` for hotspot-installer v2 users though.

As for OPDS, it lacks the human readable ID (`ekopedia_fr_all.fr`) that people are familiar with.
I understand that it's not much useful for GUI but since we have those IDs elsewhere for various reasons and because it's a practical method on CLI, it might be interesting to integrate it in the feed?

Also, it seems like the output of OPDS is not gziped. Might be a good addition.

rgaudin, 2020-01-16:
In GitLab by @Kelson42 on May 10, 2019, 08:55

@rgaudin thank you for reporting the problem with HTTP compression on library.kiwix.org. I have reported it properly here https://github.com/kiwix/maintenance/issues/57.

@rgaudin for the "human readable ID", please open a ticket in kiwix-tools repo... I'm not so sure this is a good idea myself, but let see what other poeple have to say about this idea.

kelson42, 2024-06-04:
Implemented in imager-service