offspot/kiwix-hotspot #27: [USABILITY] Log window vertical scrollbar should be able to follow

rgaudin, 2020-01-16:
In GitLab by @Kelson42 on Jul 1, 2017, 11:31

If the vertical scrollbar displays the last lines of the log (which is the case per default at the beginning), it should follow the log and always display the last lines. This should be the case as long as nobody has changed it back to display something earlier. If the scrollbar comes back to the end, then it should again follow automatically

rgaudin, 2020-01-16:
In GitLab by @thiolliere on Jul 3, 2017, 15:42

partially done in f083c80eb7e655ffa504a2aaa5f9ade583569458

still have to not scroll when user scroll up

rgaudin, 2020-01-16:
In GitLab by @Kelson42 on Jul 4, 2017, 18:25

Following the last lines seems to not work properly for me, the last lines are not always visible.

kelson42, 2024-06-04:
Unfortunately this product has been discontinued. Therefore this issue won't be fixed/implemented. A bit more details can be found [here](https://github.com/offspot/kiwix-hotspot/blob/main/README.md).

closed 2024-06-04 by kelson42