offspot/kiwix-hotspot #48: Be sure that pibox-installer is translatable.

rgaudin, 2020-01-16:
In GitLab by @mgautierfr on Jul 7, 2017, 09:27

Not a ultra priority (fixing functionnal bug is more important), but we should be sure that pibox-installer interface is translatable.

This means (at least) :
- integrate gettext functionality in python when necessary.
- Check that glade definition file are translatable (and how).

rgaudin, 2020-01-16:
In GitLab by @bochecha on Jul 7, 2017, 09:35

> Check that glade definition file are translatable (and how).

That's relatively easy, you just need to set the `translatable` property of a widget to `yes`. ([example](https://git.gnome.org/browse/totem/tree/data/preferences.ui#n13))

Modern versions of gettext can extract the strings normally when they are marked this way.

rgaudin, 2020-01-16:
In GitLab by @Popolechien on Aug 2, 2019, 15:27

Yep, I confirm that it would be nice. Low priority, but nice. At this stage the language selection has some weird-ass options (bambara, amharic, etc.), so expanding to more demanded options would be nice and, considering the limited number of strings involved, fairly easy to implement I'm sure.

rgaudin, 2020-01-16:
In GitLab by @rgaudin on Aug 4, 2019, 11:35

Well noted but let me still remind here that the current language list is French/English.

[image]

kelson42, 2024-06-04:
Unfortunately this product has been discontinued. Therefore this issue won't be fixed/implemented. A bit more details can be found [here](https://github.com/offspot/kiwix-hotspot/blob/main/README.md).

closed 2024-06-04 by kelson42