offspot/kiwix-hotspot #65: [USABILITY] Windows firewall complains about qemu

rgaudin, 2020-01-16:
In GitLab by @Kelson42 on Jul 14, 2017, 05:54

This is annoying. I'm not even sure it blocks concretly something. We should try to find a way to avoid that.

rgaudin, 2020-01-16:
In GitLab by @rgaudin on Mar 7, 2018, 17:57

We can't remove this dialog completely but we can restrict it to only the first use by setting the MAC address of the network card.

Would that be satisfying?

Ex: `ip link set dev tapn address 52:54:00:12:34:56`. See [Qemu Networking](https://en.wikibooks.org/wiki/QEMU/Networking)

rgaudin, 2020-01-16:
In GitLab by @Kelson42 on Mar 7, 2018, 17:58

OK

rgaudin, 2020-01-16:
In GitLab by @Kelson42 on Nov 18, 2018, 17:07

@rgaudin I'm not sure about the status of this... I do not have seen this windows anymore...

rgaudin, 2020-01-16:
In GitLab by @rgaudin on Nov 18, 2018, 17:36

It's still there, it's only visible once if you tick the appropriate box ; that's why you don't see it anymore.

kelson42, 2024-06-04:
Unfortunately this product has been discontinued. Therefore this issue won't be fixed/implemented. A bit more details can be found [here](https://github.com/offspot/kiwix-hotspot/blob/main/README.md).

closed 2024-06-04 by kelson42