offspot/kiwix-hotspot #193: Nightlies and official releases should deal properly with base images

rgaudin, 2020-01-17:
In GitLab by @Kelson42 on May 11, 2018, 08:11

* Official releases of the images should be published beside pibox-installer binaires - automatically
* Nightlies should build and use nightly base-image automatically.

rgaudin, 2020-01-17:
In GitLab by @rgaudin on May 11, 2018, 09:10

> Official releases of the images should be published beside pibox-installer binaires - automatically

Agrees. We'll just need to add the SSH private key to the repo's settings.

> Nightlies should build and use nightly base-image automatically.

Makes sense but we'll have to dynamically change hard-coded values.

rgaudin, 2020-01-17:
In GitLab by @Kelson42 on Jun 10, 2018, 12:24

@rgaudin Can we do that now? Or do we have additional/other problems?

rgaudin, 2020-01-17:
In GitLab by @rgaudin on Jun 12, 2018, 08:41

Well, a few things needs to be sorted out first:

**Access**

Base image can only be built on framagit as it takes about 5h and travis would timeout before.

So we need the private key of `repos.ideascube.org` in the framagit settings (@mgautierfr)

We could use `download.kiwix.org` as well since it's already a requirement for many files.

**What is an official release of the base image?**

I guess it's a (base-image) build launched on the master branch. Is that sufficient? Do we want to limit that to a tag or something?

Since we have not yet merged to master, we can't just-do-it at the moment.

**Updating code**

Changing the base image on the code is as easy as editing the [contents.json](https://framagit.org/ideascube/pibox-installer/blob/image-creator/contents.json) which can be done automatically on new-build success

``` json
    "pibox_base_image": {
        "url": "{mirror}/plug/base/pibox-master_2018-06-10.img.zip",
        "name": "pibox-master_2018-06-10.img.zip",
        "root_partition_size": 7000000000,
        "checksum": "21e6b5416c7e8112f01c7584a52cda41caa4842c96d6e7e5f4679ced778f0b8c",
        "archive_size": 2965013489,
        "expanded_size": 8000000000
    },
```

* We could easily rebuild a `pibox-installer` right after the image (we already do it [just before](https://framagit.org/ideascube/pibox-installer/blob/image-creator/.gitlab-ci.yml)) but that would be limited to linux.
* macOS and Windows are on different platforms (travis and appveyor) and that would require a commit to trigger those.
* Do we want to trigger a commit from the CI ???

rgaudin, 2020-01-17:
In GitLab by @Kelson42 on Nov 18, 2018, 17:34

It is a bit sad that the creation/maintaining of these base image is not as clear as the soft itself.

kelson42, 2021-04-21:
@rgaudin I believe I have changed my opinion on this ticket. Here like for any other software, I’m not in favour of having a nighkty build using a “random” latest dep version of a dependence. This, whatever if this is an internal dependence or an external. Therefore I believe this ticket should be a lot easier to implement: we just need to publish via CI the master images and these images have to be tagged. Ideally the master images tags should be the same as the kiwix hotspot tags, but then we get a chicken egg problem. So we should probably have a dedicated tag set for the master umages. Does that make sense?

rgaudin, 2021-04-21:
It does but here's what I propose:
- master images created in CI as were previously (once a month).
- In addition to date in master image filename, we add the commit short ID
- master creation workflow updated to skip master creation if master-related files not touched since last master success. It sounds a bit overkill but I think it's gonna be really useful.
- whenever we release an hotspot version we also tag that commit with a specific tag (`base-image-v2.4.0`). Same commit could thus be tagged multiple times if base image didn't change.

kelson42, 2021-04-21:
> * master images created in CI as were previously (once a month).

Why? I don't understand the value of doing so (in comparison to just create a tag and doing a release on demand on if needed) and we will create and have to store many masters and only a small subset of them will be needed.

> * In addition to date in master image filename, we add the commit short ID

Why? Is that because the image does not match a tag and therefore we need to know exactly which version of the code is inside?

> * master creation workflow updated to skip master creation if master-related files not touched since last master success. It sounds a bit overkill but I think it's gonna be really useful.

I understand the need to avoid creating a lot of duplicate master images, but we don't need such a subtil behaviour if we just create/publish a master image if requested?

> * whenever we release an hotspot version we also tag that commit with a specific tag (`base-image-v2.4.0`). Same commit could thus be tagged multiple times if base image didn't change.

Why having a tag if we publish new master image all months?

Remark: It seems to be that part of the master images would be uploaded as nightly and other as release, but this is not clear from your comment. It suspect as well that you want to use the CI to compile and publish test master images as nightly, to then be tested by developers (that would be easier for devs). Would you be able please to confirm that I understand properly?

rgaudin, 2021-04-22:
> > * master images created in CI as were previously (once a month).
> 
> Why? I don't understand the value of doing so (in comparison to just create a tag and doing a release on demand on if needed) and we will create and have to store many masters and only a small subset of them will be needed.

Because this process is fragile and we need to check from time to time that it's still working. We don't have to upload the images though, just build them.

> > * In addition to date in master image filename, we add the commit short ID
> 
> Why? Is that because the image does not match a tag and therefore we need to know exactly which version of the code is inside?

Exactly.

> I understand the need to avoid creating a lot of duplicate master images, but we don't need such a subtil behaviour if we just create/publish a master image if requested?

It's useful to have that information (whether we'd benefit from a new master) as we don't release regularly nor frequently but it' not a need, it's just handy to have that info on GH instead of looking through history (or we could just be more explicit in changelog).

> > * whenever we release an hotspot version we also tag that commit with a specific tag (`base-image-v2.4.0`). Same commit could thus be tagged multiple times if base image didn't change.
> 
> Why having a tag if we publish new master image all months?

To easily find source code of master used in hotspot version x.y. Because tags are free

> Remark: It seems to be that part of the master images would be uploaded as nightly and other as release, but this is not clear from your comment. It suspect as well that you want to use the CI to [...]

No, definitely no as nighties. It's a resource intensive process that takes 4h. We don't to trigger that for no reason.
We can trigger it already from any branch in Github Actions. That's what we want for test builds (that we have to remove manually) and for builds that will eventually be used in a hotspot release.

commit 133707ce (2021-04-22): Fixed #193: only upload master on manual trigger

Also include commit short ID to filename
closed 2021-04-22 by rgaudin