offspot/kiwix-hotspot #261: Add version metadata to Windows binaries

rgaudin, 2020-01-17:
In GitLab by @rgaudin on Sep 20, 2018, 14:49

Due to #232, metadata for windows binaries are disabled.

Problem lies in Appveyor:
* ResourceHacker (RH) from [nuget](https://www.nuget.org/packages/ResourceHacker.CommandLineTool/) used to work properly until 09/17/2018.
* Nuget RH now runs successfully but remove UAC when overwriting VersionInfo resources.
* Portable RH version, either from [chocolatey](https://chocolatey.org/packages/reshack.portable) or [official website](http://angusj.com/resourcehacker/) fails to run on Appveyor (exit code -1, no output).
* All other tools tested  have similar issue (not only within Appveyor): they remove UAC

Tested tools:

* [rcedit](https://github.com/electron/rcedit) removes UAC ; its `--set-requestExecutionLevel requireAdministrator` option does not work.
* [StampVer](http://blog.dixo.net/downloads/stampver/) can only update version number and those must match original binary (not applicable for us).
* [VerPatch](https://www.codeproject.com/KB/install/VerPatch.aspx?msg=3207401) removes UAC

rgaudin, 2020-01-17:
In GitLab by @rgaudin on Sep 20, 2018, 17:52

On Windows Server 2016 (Visual Studio 2017 image), portable RH does not crash but still removes UAC.

kelson42, 2021-04-20:
@rgaudin Now that we are using Github actions, is this feature doable?

commit b833a04d (2021-04-20): Fixed #261: Update windows build metadata

Now manually adding UAC manifest  as otherwise removed by resource hacker

- using a kiwix copy of resource hacker as official website is not versioned
- updating resource hacker resource with proper version in CI
- using a kiwix copy of qemu installer as source repo is sometimes slow
closed 2021-04-20 by rgaudin