offspot/kiwix-hotspot #275: Add Batman-adv mesh support

rgaudin, 2020-01-17:
In GitLab by @sam_uk on Oct 21, 2018, 13:40

Could you run these scripts?

https://github.com/chootka/subnodes-lighttpd

It would mean that the hotspot's functional area could be easily extended at a cost of around £10 per additional node (Pi Zero + cheap USB wifi dongle)

rgaudin, 2020-01-17:
In GitLab by @Kelson42 on Oct 21, 2018, 14:59

@sam\_uk I challenge the idea of the mesh for Kiwix hotspot. Regarding the cheap price of each hotspot, is that not just easier to setup N RPIs with the same content?

rgaudin, 2020-01-17:
In GitLab by @sam_uk on Oct 21, 2018, 16:09

Hi Kelson,

So user story might be; large school buys 
RPI B+ (£35) they get a 120Gb  mSata card (£25), mSata adaptor (£8). They spend four days of staff time downloading and curating content in the nearby city.

It's a huge success! But only 1/3 of classrooms are in range.

They either spend a further £136 and eight additional staff days, building two new units, or they spend £25 on two Zeroes with external wifi dongle.

rgaudin, 2020-01-17:
In GitLab by @Kelson42 on Oct 21, 2018, 17:21

Let assume the teacher has one RPI and one configured microSD card, this is what he has to do/achieve anyway.

What I tend to say is: buy this hardware (RPI+SD) as many times (3x in your example) as you need and just clone your configured SD card on the other cards. I agree that this might be more expensive (in $) than having additional RPIs zero but:
* This does not need any additional time (just clone the SD)
* This is easier to understand and deploy
* This is robust: one might fail and you still have a backup
* This will be adapted at the usage. ie. if you need 3x RPI hotspots to cover the classroom, then I suspect this is a quite bit classroom with probably more than 30 students. 1 RPI is anyway probably not strong enough (CPU/RAM) to deal properly with so many users.

All that said I'm not at all against the feature request (support mesh), but I strongly suspect this won't really work as expected if you have so many students.

kelson42, 2022-10-30:
I guess this is a ticket for `base-image`?

rgaudin, 2022-10-30:
Yes but I think it's not relevant anymore. There's been a single user request in 4 years and the use case can be implemented differently.

We all agree it would be nice to have mesh support but this means a lot of implenting work, a lot of testing work and more importantly a hell to assist users with.
On the other end, we can advise uses to plug their Pi to a stronger WiFi AP device and/or use dedicated WiFi repeaters with zero maintenance nor implementing cost for us yet probably a better result.

Should there have been a lot of requests, we could have investigated cheaper solutions for users but at the moment, I'd vote for simply closing this ticket.

kelson42, 2022-10-30:
@rgaudin Agree

closed 2022-10-30 by kelson42