offspot/kiwix-hotspot #330: MacOS version should follow MacOS UI guidelines

rgaudin, 2020-01-17:
In GitLab by @Kelson42 on Nov 18, 2018, 17:37

This can probably be done by changing the UI framework from PyGTK to Qt or Electron

rgaudin, 2020-01-17:
In GitLab by @rgaudin on Nov 18, 2018, 17:41

hum, it's a lot more than just switching framework but… some have better/easier OS integration for basic stuff like Menu bar, Preferences, etc.

Agrees we should share all OS integration issues inside this one.

kelson42, 2020-07-07:
To me this is not really a bug because it works as designed. It is just annoying.

rgaudin, 2020-07-09:
Electron alternative: https://tauri.studio/

kelson42, 2024-06-04:
Unfortunately this product has been discontinued. Therefore this issue won't be fixed/implemented. A bit more details can be found [here](https://github.com/offspot/kiwix-hotspot/blob/main/README.md).

commit 307e23d1 (2019-10-09): Fixed #330: added Nomad Android App content feature
commit 3229c5ea (2019-10-09): Merge branch 'nomad' into 'master'

Fixed #330: added Nomad Android App content feature

Closes #330

See merge request ideascube/kiwix-hotspot!59
closed 2024-06-04 by kelson42