offspot/kiwix-hotspot #361: Add button " copy log to file"

rgaudin, 2020-01-20:
In GitLab by @Kelson42 on Dec 2, 2018, 08:57

To quickly be able to save the log in a file

rgaudin, 2020-01-20:
In GitLab by @Kelson42 on Dec 12, 2018, 08:08

or maybe this should replace the current one?

rgaudin, 2020-01-20:
In GitLab by @rgaudin on Dec 12, 2018, 08:10

What I'd like in a future release is a simple steps completion progress that any user could read with a potential error message on failure and the full log to be written on disk for submission.

kelson42, 2024-06-04:
Unfortunately this product has been discontinued. Therefore this issue won't be fixed/implemented. A bit more details can be found [here](https://github.com/offspot/kiwix-hotspot/blob/main/README.md).

closed 2024-06-04 by kelson42