offspot/kiwix-hotspot #408: Replace EDUPI with a more standard solution

rgaudin, 2020-01-20:
In GitLab by @Kelson42 on Jul 4, 2019, 07:52

EDUPI software solution is not maintained and I'm convinced we can find a standard solution - easier to maintain and deploy.

rgaudin, 2020-01-20:
In GitLab by @Kelson42 on Nov 20, 2019, 12:19

Consider https://afterlogic.org/aurora-files

kelson42, 2022-04-05:
An other alternative https://peergos.org/

rgaudin, 2022-04-06:
FYI doesn't offer a docker image. Is in Java.

kelson42, 2022-06-04:
@rgaudin @Popolechien We will have to make a choice in the next weeks if we want to avoid to Dockerize Edupit for the new image-creator. After exploring a bit to topic again, I have found https://www.files.gallery/ which seems to be the best so far. Should we go with it?

kelson42, 2022-06-04:
Unfortunately files.gallery is not free and quite CDN dependant. But the base is a simple PHP file and we are allowed to use it how we want. So I would suggest to still consider using it, either by modifying it a bit or take contact with the author and see if something can be done with him.

An other product worth a look is http://bpatrik.github.io/pigallery2/

rgaudin, 2022-06-09:
Just to be clear, except the last one, those are new feature requests that may be outside the scope of such a tool.

rgaudin, 2022-07-12:
@kelson42 files.gallery is [on Github](https://github.com/mjau-mjau/files.photo.gallery) but it expressively prevents us from doing what you suggest.

[pupcloud](https://pupcloud.vercel.app) (https://github.com/proofrock/pupcloud) might be a good candidate. It's UI is limited but the basic features are present: files browsing, displays pictures, videos, audios and PDF, allows file downloads and uploads, responsive. It is also lightweight and and is a single binary (go) with an existing dockerfile.

kelson42, 2022-10-16:
I believe it would be better to replace this ticket to the `Edupi` repository

rgaudin, 2022-10-17:
To the images one I guess ; created https://github.com/offspot/container-images/issues/11

closed 2022-10-17 by rgaudin