offspot/kiwix-hotspot #416: Use new tagging system

rgaudin, 2020-01-20:
In GitLab by @rgaudin on Aug 27, 2019, 15:12

In order to support the [new tagging system](https://wiki.openzim.org/wiki/Tags), please make sure your reader:

* is feeding itself from the OPDS stream (using a local cache/copy or not)
* is using the lastest libkiwix (not released ATM). It supports new options and includes the backward-ompatibility bits you'd need.
* is using the [centralized Categories List](https://github.com/kiwix/kiwix-tools/issues/317)
* is using the new tags only for flavours and full-text index.

**NOTE**: This issue for planning purposes, OPDS and libkiwix **does not support** those features at the moment.

kelson42, 2021-12-03:
We should do that at the time we use OPDS.

kelson42, 2024-06-04:
Unfortunately this product has been discontinued. Therefore this issue won't be fixed/implemented. A bit more details can be found [here](https://github.com/offspot/kiwix-hotspot/blob/main/README.md).

closed 2024-06-04 by kelson42