offspot/kiwix-hotspot #419: Add language codes to dialogs with more than one language

rgaudin, 2020-01-20:
In GitLab by @frief on Sep 8, 2019, 17:08

please add the language codes to dialogues and selections where multiple languages are involved.

Without these the selection of the intended zim file might need to be guessed.

Appending three files:
* a screenshot of the mobile phone which shows the difficulty users might be facing[image]
* a screenshot of the installer. Language codes in the left dialog and on the bottom right would be very helpful. (I ended up editing the configuration file)
[image]
* the configuration file to reproduce it.
[kiwix_hotspot_test.json](/uploads/a3227667f9fee03b0fa725a9b13ad35e/kiwix_hotspot_test.json)

Thanks for Kiwix!

rgaudin, 2020-01-20:
In GitLab by @Kelson42 on Sep 8, 2019, 18:40

We have here two problems (because of similar content metadata) :
* Possible confusion in Kiwix Hotspot
* Possible confusion in Kiwix Hotspot installer

In Version 3.0 of Kiwix Hostpot they should be unified (same look & feel in Kiwix Hotspot and Kiwix Hotspot Installer), so hopefully this will become only one problem. See #269.

#269 is also about allowing users to customise their cards title/description...

@frief Would that be enough to your opinion to improve the usability?

rgaudin, 2020-01-20:
In GitLab by @frief on Sep 8, 2019, 21:53

Theoretically yes. The bottleneck is that I do not know most of the languages that this (somewhat wiktionary centered) configuration uses.

The most I could add to content of the *Welcome Page* would be some English text and/or the language codes.
As English is not the first language of the target audience (and not my native language either) it's probably best to just display the language code.

I was hoping for an algorithm like:

*IF (the mandatory ISO639-3 language identifiers of the included ZIM files are not reducible to a common denominator) THEN { additionally display the language identifier of all ZIM files }*

rgaudin, 2020-01-20:
In GitLab by @Kelson42 on Sep 9, 2019, 01:59

@frief Thank you for the clarification
@rgaudin Not sure I want to follow the path for the use case where the SD card creator does not understand the language(s) of the users...

kelson42, 2024-06-04:
Unfortunately this product has been discontinued. Therefore this issue won't be fixed/implemented. A bit more details can be found [here](https://github.com/offspot/kiwix-hotspot/blob/main/README.md).

closed 2024-06-04 by kelson42