offspot/kiwix-hotspot #422: Installer should not accept logo files that are not images

rgaudin, 2020-01-20:
In GitLab by @Popolechien on Oct 28, 2019, 09:05

Someone created a Configuration on the cardshop using a PDF file as source for the logo - it obviously did not work (see screenshot below), but the installer nonetheless accepted it and the user could not know until they downloaded, installed and accessed the hotspot. I suggest either limiting the formats acceptable to .png or .jpg or at the very least change the wording (both on the installer and cardshop to "jpg/png format only".
 
[image]

rgaudin, 2020-01-20:
In GitLab by @Popolechien on Oct 28, 2019, 09:15

changed the description

kelson42, 2024-06-04:
Unfortunately this product has been discontinued. Therefore this issue won't be fixed/implemented. A bit more details can be found [here](https://github.com/offspot/kiwix-hotspot/blob/main/README.md).

closed 2024-06-04 by kelson42