offspot/kiwix-hotspot #423: Force logo refresh on captive portal

rgaudin, 2020-01-20:
In GitLab by @Popolechien on Oct 28, 2019, 13:23

So I ran two images on my Raspberry, the first one from Orange Madagascar and the second a simple "Test", with the default Kiwix logo. Yet on the captive portal I still get to see the Orange logo (then moving on to test.hotspot everything is normal and I see the expected Kiwix logo):
[image]

kelson42, 2021-04-21:
@Popolechien Actually this problem can occur not only with the logo but with any content having the same URL on two differrent Hotspots.
@rgaudin IMO, we should add something like ?randomid to this URLs.

rgaudin, 2021-04-21:
> @Popolechien Actually this problem can occur not only with the logo but with any content having the same URL on two differrent Hotspots.

No, only on different hotspots with the same `name` which limits it and also I've never encountered it with Firefox. I think this issue exists because the captive portal is displayed inside a system-controlled webview with specific-and-unknown cache settings.

> @rgaudin IMO, we should add something like ?randomid to this URLs.

Absolutely, we should do that for the captive portal as the URL is the same for all hotspots.

Please note though that it affects a negligible fraction of users: those accessing multiple hotspots, from a mac/ios.