offspot/kiwix-hotspot #424: Make hotspot directly pluggable onto screen

rgaudin, 2020-01-20:
In GitLab by @Popolechien on Nov 1, 2019, 10:31

So we have a use case where everyone in the room has a tablet or smartphone but the teacher also wants to showcase on a large screen (projector or TV screen). Ideally, we could/should be able to simply plug the hotspot onto it and have visual display.

rgaudin, 2020-01-20:
In GitLab by @Kelson42 on Nov 1, 2019, 10:39

This should be possible to do so by having a kind of kiosk mode starting a browser in  fullscreen. The challenge is probably to be able to implement this feature without disadvantaging too much the other users (more storage and more memory used).

kelson42, 2020-05-05:
Someone made something around these lines and hase given a few hints:
* https://www.youtube.com/watch?v=gtlydT6Lumg&feature=youtu.be
* https://twitter.com/rickmakes/status/1257479726211481600
* https://www.rickmakes.com/locked-down-offline-wikipedia-kiosk-on-raspberry-pi-4/

It seems that the solution works with `surf-display` https://manpages.ubuntu.com/manpages/cosmic/man1/surf-display.1.html

kelson42, 2021-04-20:
If we implement this, we shoukd probably launch it only if something connected  via HDMI.

rgaudin, 2021-04-20:
So is this feature request accepted?

kelson42, 2021-04-20:
Not sure on my side

Popolechien, 2021-04-20:
Well considering the amount of people who wrongfully do this (despite the necessary how-to, user manual, FAQ, etc.), I'd say that we'll soon reach the point where we either need to set up an automated message or deal with the issue. There's also been the request made in the context of training sessions (e.g. show on screen what everyone in the room should see on their device)

Bottom line: not urgent, can wait 2.5, but I would want to see it eventually.

rgaudin, 2021-04-20:
I'm also in favor. Unless there's a huge downside (consumes too much resources all the time, takes too much space), it makes very little sense to prevent it.

kelson42, 2021-04-20:
My concerns are moderate but:
* Memory consumption, but probably problem solved if launched only via hdmi trigger
* Master partition disk usage, it is already quite big
* No idea of the complexity/robustness of the solution. But if only a few days of work, then probably OK.

kelson42, 2023-08-29:
@rgaudin Should we move this ticket to `base-image`?

kelson42, 2023-10-29:
> No we already have https://github.com/offspot/base-image/issues/51

It's two different things.

> It's been agreed that we won't support this in the forseeable future because we can't conditionally enable it and we don't want all the stack.

I don't remember that we can't conditionally enable it, do we have a written trace of that? Actually are we sure that xrandr can not do it? Whatever, having this feature activated conditionaly seems to me to be a detail. It can perfectly be an option an image creation time to me.

> We could imagine documenting (or providing a script) to properly install/configure such a desktop on top of an image of ours, but that would not be at base-image anyway

That would be for sure a good, maybe even necessary step, if we decide to invest time on this feature.

rgaudin, 2023-10-30:
> I don't remember that we can't conditionally enable it, do we have a written trace of that? Actually are we sure that xrandr can not do it? Whatever, having this feature activated conditionaly seems [...]

We are not launching the system anymore. That's the whole point of the switch. It thus leaves us with a limited set of options:
- Include a complete Desktop environment in the _base image_ (otherwise it's not useful). Whether this is running on start can be conditional but it's installed/maintained anyway. We've discussed that in the past and agreed we don't want that.
- Maintain two base images: a desktop-enabled one and a regular one.
- Hack an install on first boot: gather packages for the DE, add them to storage as an option and on first start, install that ton of debs.

All those options add a **big** maintenance burden. The alternative is instructions providing the few apt commands to launch on a live –**connected**– system.

> > We could imagine documenting (or providing a script) to properly install/configure such a desktop on top of an image of ours, but that would not be at base-image anyway
> 
> That would be for sure a good, maybe even necessary step, if we decide to invest time on this feature.

👍

benoit74, 2024-01-26:
Potential solution https://github.com/balena-labs-projects/browser