offspot/kiwix-hotspot #428: Chinese phones in endless reboot when connected to Kiwix hotspot

rgaudin, 2020-01-20:
In GitLab by @Gabe1290 on Nov 12, 2019, 15:45

During a trip to Senegal in October I found out that certain budget chinese phones (Samsung clones) would go into an endless reboot loop once connected to a Kiwix Hotspot.
This would happen with Raspberry W and Raspberry 4 using a 128 GB SD card, 64 GB and 32 GB.
The only solution was to turn off the Raspberry, in one case I witnessed it was not possible to switch off the phone.

This problem did not occur when being connected to WikiMed hotspot distributed by James Heilman.

I also have chinese phones, HomTom models, and did not have this problem. But a least 4 senegalese teachers were not able to use the Kiwix Hot spot.

My main contact in Senegal, Ibrahima Diop, has one of these problematic chinese phones.

For more details about the phones used, we will have to contact the Senegalese partners.

Gabriel

rgaudin, 2020-01-20:
In GitLab by @rgaudin on Nov 12, 2019, 15:49

Which version of the hotspot ? You can see it at
http://version.xxxx.hotspot/

Can you elaborate a bit on the symptoms ? How would one try to reproduce it
?
You just connect to its WiFi network and the phone immediately reboots
forever ?
Did the hotspot had a wifi password ?

Thanks for the report.

rgaudin, 2020-01-20:
In GitLab by @Gabe1290 on Nov 12, 2019, 16:14

I left  all the hotspots (about 20 of them) in Senegal, I will have to flash one to see the version. I downloaded the program around October 10 or 11

No symptoms: you connect the phone to the Kiwix hotspot and it starts rebooting. Almost straight away, every time.

No wifi password.

rgaudin, 2020-01-20:
In GitLab by @Gabe1290 on Nov 12, 2019, 16:15

I guess we need one of those budget phones...

rgaudin, 2020-01-20:
In GitLab by @Kelson42 on Nov 13, 2019, 22:54

@Gabe1290 Thank you for finally use make Kiwix Hotspot and have made a bug report. We are committed to fix problem, but without being able to reproduce the problem this will be hard. Like you can imagine, none of us has ever faced that problem. It is really important to get more details about the phone and what happens exactly.

rgaudin, 2020-01-20:
In GitLab by @Kelson42 on Nov 22, 2019, 14:47

One thing I have not said: if the device reboots then this is definitely a bug in the  device/OS/browser. That said we might be able to do something on Kiwix Hotspot side to avoid to trigger that bug. But, without additional information allowing to reproduce the problem we will probably have to close that ticket without any solution.

rgaudin, 2020-01-20:
In GitLab by @Kelson42 on Nov 23, 2019, 16:24

Problematic devices are:
* Samsung Galaxy S9
* Techno k7

@rgaudin Do you know by any chance someone having an S9

kelson42, 2020-04-18:
@rgaudin What would be the next steps here?

rgaudin, 2020-04-18:
get our hands on one of those

rgaudin, 2021-04-15:
Sorry this has gotten quite old. Please re-test with 2.4.0 release and reopen if you're still experiencing it.

closed 2021-04-15 by rgaudin