offspot/kiwix-hotspot #429: Prebuilt system of master does not scale

rgaudin, 2020-01-20:
In GitLab by @Kelson42 on Nov 20, 2019, 11:35

Ce current master takes 7GB. If we add support for X, Moodle, Kolibri, etc... et will become bigger and bigger. This can be quite a waste of storage if all of this is not needed by the user.

We have chosen the system because this is quick to install and really reliable, which is great... but maybe we could find a better alternative where we could keep these advantages without potentially losing so much mass storage?

kelson42, 2020-04-19:
I keep thinking about that topic.

The fundamental problem here is that we would need a bunch of precomputed master images coverning all scenarios... but this is not doable because the number of possible master images is just too high.

But maybe we could kind of hybrid the flexibility of the original approach (each build installs only what is necessary) and the power of the most recent one (we have a fix master). We could move from a fix master to many masters which are managed like cached versions.

That means that a card creation would generate using ansible the master needed or use one in cache if possible. It could be exactly what is needed or maybe a master image which does a bit more. If not it would create one and cache it for next usage.

We might also think about delegating this task to the cloud (never locally in kiwix-hotspot installer).

kelson42, 2021-05-08:
Latest status is that we shoukd be able to run Dockerized apps on the RPI. We know now how to publish Docker images for ARM. The RPIs are meanwhile more powerful. We might use K8s to start/stop smartly containers tomsave resources.

If we achieve to install Docker images and create containers at run time, we won’t probably need qemu anymore. That means that the card creation process will be simpler and faster, which means that we don’t need a master with all the apps anymore. In other words, we could have a system partition with a variable size.

one difficulty left is to install the Docker images without using the Docker local deamon (which miht be cumbersome to run in the installer). To achieve to do that, we could snapshot the images in tarball, copy them in the system partition at install and load/import then at first run.

The containers would be created on demand if we find a way to configure k8s that way, based on reverse proxy request. That approach would slightly slow down the first request, force-us to properly mount the right paths to save persistent data, but all the rest (like logs) would be removed often.

All of this would not only allow more flexibility for the master, but as well, will speed-up the image creation.

kelson42, 2021-05-26:
Smaller versions of K8s (k3s, microk8s) still need 512MB for the central node and latest version don’t support anymore armv6. For these two reason, such an approach won’t run on RPI 0 W.

kelson42, 2022-06-08:
Closing, work on the container base new creator in repository `offspot/base-image` and `offspot/image-creator`.

closed 2022-06-08 by kelson42