offspot/kiwix-hotspot #430: Not optimal captive workflow on Windows 10

rgaudin, 2020-01-20:
In GitLab by @Kelson42 on Nov 22, 2019, 10:38

Steps to reproduce:
* Connect to Kiwix Hotspot Wifi Hotspot
* Windows ask you which browser you want to use
* I choose Firefox
* Captive portal pages are displayed and they basically tell me to open "http://kiwix.hotspot" in a browser

What I want:
Considering that I'm already in Firefox (but that works for any standard browser), this message does not make sense to me. I want to be directly redirected to the Kiwix Hotspot welcome page (http://kiwix.hotspot)

rgaudin, 2020-01-20:
In GitLab by @Kelson42 on Nov 22, 2019, 10:39

changed the description

rgaudin, 2020-01-20:
In GitLab by @rgaudin on Nov 22, 2019, 10:49

assigned to @rgaudin

rgaudin, 2020-01-20:
In GitLab by @rgaudin on Nov 22, 2019, 10:49

Right. we should change this based on User Agent.

I think I recall a case where the captive window fakes a real browser (maybe it was Safari on Mac).

kelson42, 2021-12-04:
@rgaudin I wonder how valid/updated is this ticket consodering that you have recently checked this stuff for Windows11?

rgaudin, 2021-12-04:
haven't checked win10 in this round