offspot/kiwix-hotspot #503: Does not run on Debian 10

teran-mckinney, 2020-03-19:
```
$ ./kiwix-hotspot 
Traceback (most recent call last):
  File "site-packages/PyInstaller/loader/rthooks/pyi_rth_pkgres.py", line 13, in <module>
  File "/home/travis/build/kiwix/kiwix-hotspot/hotspot-env/lib/python3.7/site-packages/PyInstaller/loader/pyimod03_importers.py", line 623, in exec_module
  File "site-packages/pkg_resources/__init__.py", line 86, in <module>
ModuleNotFoundError: No module named 'pkg_resources.py2_warn'
[8868] Failed to execute script pyi_rth_pkgres
```

Not really sure how this is packaged and how it's supposed to be working.

kelson42, 2020-03-19:
@teran-mckinney  Works for me on a Ubuntu x86_64 20.04 (based on Buster). Can you please:
* Check that you have downloaded/run https://download.kiwix.org/release/kiwix-hotspot/v2.3.2/kiwix-hotspot-linux.tar.gz
* Give details about your CPU architecture?

teran-mckinney, 2020-03-19:
Hmm, looks different than what I got. I downloaded from the same link and got a different checksum the first time around.

Do you have multiple mirrors in rotation?

I'm on x86_64, by the way.

```
$ sha256sum kiwix-hotspot-linux.tar.gz 
b8623db32d1736e52745f7ede15059400b07aa922ef2d8572188aec40e5266d5  kiwix-hotspot-linux.tar.gz
```

(Working.)

```
$ sha256sum kiwix-hotspot-linux.tar.gz 
f18b52d24481559191b9fb10ba11a0c62780bb26933f018949c7e1c1df882820  kiwix-hotspot-linux.tar.gz
```
(Not working.)

Thank you for your help.

kelson42, 2020-04-18:
@teran-mckinney Sorry for the late answer. Yes we have mirrors but they should be updated.

closed 2020-04-18 by kelson42