offspot/kiwix-hotspot #504: Update logo

Popolechien, 2020-03-22:
There is a new, slightly remodeled logo for Kiwix 
https://commons.wikimedia.org/wiki/File:Kiwix_logo_v3.png
The app's icon needs to be updated accordingly

DhairyaBahl, 2020-10-05:
I would like to work on this issue.

rgaudin, 2020-10-05:
@DhairyaBahl OK. Please consider comments from #530. Ideally, tackle both together.

Popolechien, 2020-10-05:
Ah, comments in #530 are in French. I'll clarify in English in a minute.

DhairyaBahl, 2020-10-05:
@Popolechien https://commons.wikimedia.org/wiki/File:Kiwix_logo_v3.png this link in the description of the issue is showing error that this page is deleted.

Popolechien, 2020-10-05:
Ah yes, we posted an SVG instead. Here it is: 
https://commons.wikimedia.org/wiki/File:Kiwix_logo_v3.svg

DhairyaBahl, 2020-10-05:
@Popolechien Okay Got it . Thanks :)

DhairyaBahl, 2020-10-05:
@Popolechien The new logo in in svg format where as the old logo is in the format of png !! so should I should replace it i guess. There must not be any issue right ?

Popolechien, 2020-10-05:
@DhairyaBahl svg is simply easier to export into png from Wikimedia Commons (rather than the opposite). I'm not too familiar with what the default format should be in the app itself. If png worked last time let's keep using it :-)

DhairyaBahl, 2020-10-05:
@Popolechien Okay, thanks a lot for guiding. :)

rgaudin, 2021-04-20:
Done in #569

closed 2021-04-20 by rgaudin