offspot/kiwix-hotspot #505: Replace "Register/S'enregistrer" with "Confirm/Compris!" on captive portal

Popolechien, 2020-04-14:
Register specifically means that some personal data will have to be entered, which is not the case. We mostly want the user to confirm the understood the instructions.

rgaudin, 2020-04-14:
OK but FYI, register was chosen exactly because we record on the device that the user (mac address, IP address, browser) requested to get in. And we keep that info for a period of time. That's how the captive portal works.

Doesn't need to reflect on UI wording though.

kelson42, 2021-05-08:
@Popolechien Any new comment?

kelson42, 2021-05-08:
@rgaudin @kelson42 Please find an agreement so we can close this ticket please.

rgaudin, 2021-05-08:
Pretty sure it's already "Agree and continue to kiwix.hotspot".
Will check

closed 2021-05-10 by rgaudin