offspot/kiwix-hotspot #507: Remove ZIM and other labels from landing page display

Popolechien, 2020-04-23:
On the hotspot's content listing page there is a lable to indicate zim or other types of file; these have no real informational value and are thus a bit confusing. They should / could be removed with no loss nor regret.

[image]

kelson42, 2020-04-28:
@Popolechien We should think to an alternative setup for these card? Would you have a mockup to propose?

rgaudin, 2020-04-28:
We removed them on the campus because it doesn't have much value. It's something we copied from ideascube but given the number of apps we have and the fact that all ZIMs are categorized the same, it's value is close to zero and raises more questions than it solves.

kelson42, 2021-04-21:
Ok, we shoukd just do it, trivial.

commit 6b77c77b (2021-04-21): Fixed #507: removed labels
closed 2021-04-21 by rgaudin