offspot/kiwix-hotspot #511: Internal Server error

18Markus1984, 2020-06-21:
Hello I wanted to create a small RaspberryPi KiWix Server from a recent YouTube Video. For this project I used my RB 3b+. I created the image and saved it to my storage. After that I wrote the image to sd with Etcher. Then I powered the RB 3b+. A hotspot was created and I also can enter "http://apocpi.hotspot" , but the only thing on this side is internal server error. Maybe you can tell me the prefered version to avoid this bug. I used the Version 2.3.
I'm happy to hear from you.

kelson42, 2020-06-21:
@rgaudin Looks like a bug, but can not really judge.
@18Markus1984 Have you saved your configuration. Would you be able to share it here please?

18Markus1984, 2020-06-21:
I didn't save the configaration. I have the image files, because I tried it two times with the same bug. But I can repate erverything and save the logs ;)

kelson42, 2020-06-21:
@18Markus1984 A developer will probably answer to you tomorrow, but IMO what would be really helpful would be:
* Securing you are using https://download.kiwix.org/release/kiwix-hotspot/v2.3.2/ (latest release)
* Make a new run (saving the configuration)
* Saving the log.

With that, if the bug is reproducable and confirmed, I hardly see how we should not be able to provide a fix quickly :) Thank you again for your time and sorry for the inconvenient.

rgaudin, 2020-06-22:
Thanks for the report @18Markus1984. As @kelson42 stated, please test with latest version.
In the mean time (and with latest as well), can you please let me know the outcome of the following URL:

* http://192.168.2.1
* http://goto.kiwix.hotspot/
* http://version.apocpi.hotspot/
* http://time.apocpi.hotspot/

Also, is the captive portal working?

18Markus1984, 2020-06-22:
It's now working. I don't Know whats different. If you want I will send you all my .img files. But they are all around 100gb.

Thank you for your help ;)

closed 2020-06-22 by 18Markus1984

kelson42, 2020-06-23:
@18Markus1984 Great :) Don't hesitate to open a new ticket if you face this problem again ... or an other problem.