offspot/kiwix-hotspot #512: KIWIX fails to start on Raspberry running latest Kernel

dcoulomb60, 2020-06-23:
Hello,
I created an image of Kiwix using the Windows installer. It worked fine. Then, I burned this image to a HD, from which I should be able to boot the RPi (4B/8G). Booting from any USB device is made possible with the latest EPROM (dated 6/15/2020). I can indeed boot from USB disks containing variaous flavors of Linux for Raspberry. The problem is that the Kiwix image encounters problems with this new Kernel. Upon startup, I get messages stating:
-recovery4.elf not foune
-start4.elf : is not compatible
-USB-MSD boot requires newer software
-Get the latest software from https:://www.raspberrypi.ort/downloads/

The image contains 2 partitions: the first contains the OS, and the second contains the Kiwix files. I think that I need to know how to install the Kiwix files on an pre-existing OS. I suppose this would simply be a matter of adding some configuration files... Or it can be more complex!

Wny help would be appreciated... Thanks

Daniel Coulombe

kelson42, 2020-06-23:
@rgaudin AFAIK this is not a supported use case. You confirm?

rgaudin, 2020-06-23:
@dcoulomb60, if you're looking for Kiwix on rPi, you should download [kiwix-tools](http://download.kiwix.org/release/kiwix-tools/) for arm and you'll find `kiwix-serve` there.

This is kiwix-hotspot, a custom system for the rPi based on debian. The goal of this project being to hide all the configuration, we don't support this kind of stuff.
If you want to do it manually, look at the [ansiblecube](https://github.com/kiwix/kiwix-hotspot/tree/master/ansiblecube) folder and install/adapt from there.

closed 2020-06-23 by rgaudin