offspot/kiwix-hotspot #513: kiwix-hotspot hangs shortly after starting

mosler99, 2020-06-23:
04:10:17 PM] Getting output of ['/bin/lsblk', '/dev/sdh']
[04:10:17 PM] [STAGE 0/6: Preparations - 0%]
[04:10:17 PM] [STAGE 0/6: Preparations - 0%]
[04:10:17 PM] Preventing system from sleeping
[04:10:17 PM] Suspending xdg-screensaver
[04:10:17 PM] Calling ['/usr/bin/xdg-screensaver', 'suspend', '0x5000000']
[04:10:17 PM] Unable to disable sleep. Please do it manually.
[04:10:17 PM] --> Check System Requirements
[04:10:17 PM] Please read https://github.com/kiwix/kiwix-hotspot/wiki/System-Requirements for details
[04:10:17 PM] --> Ensure user files are present
[04:10:17 PM] --> Prepare Image file
[04:10:17 PM] Call: ['/usr/bin/udisksctl', 'dump']

and this is where it stayed. overnight even,

any ideas?
im on PopOs... if that helps

kelson42, 2020-06-23:
@mosler99 Would you be able please to share with us:
* Which version of Kiwix-Hotspot (installer) are you using? For windows/linux/macos? If this is not https://download.kiwix.org/release/kiwix-hotspot/v2.3.2/, please try again with the last version.
* your configuration of the SD card (you can save it)?

rgaudin, 2020-06-23:
Seems like PopOS is based on Ubuntu but I'm not sure.

Could you please share the output of `/usr/bin/udisksctl dump` in a terminal?

mosler99, 2020-06-23:
i am using 2.3.2, popos is based on ubuntu.

/org/freedesktop/UDisks2/Manager:
  org.freedesktop.UDisks2.Manager:
    DefaultEncryptionType:      luks1
    SupportedEncryptionTypes:   luks1
                                luks2
    SupportedFilesystems:       ext2
                                ext3
                                ext4
                                vfat
                                ntfs
                                exfat
                                xfs
                                reiserfs
                                nilfs2
                                btrfs
                                minix
                                udf
                                f2fs
                                swap
    Version:                    2.8.4

/org/freedesktop/UDisks2/block_devices/dm_2d0:
  org.freedesktop.UDisks2.Block:
    Configuration:              [('fstab', {'fsname': <b'/dev/mapper/cryptswap'>, 'dir': <b'none'>, 'type': <b'swap'>, 'opts': <b'defaults'>, 'freq': <0>, 'passno': <0>})]
    CryptoBackingDevice:        '/'
    Device:                     /dev/dm-0
    DeviceNumber:               64768
    Drive:                      '/'
    HintAuto:                   false
    HintIconName:               
    HintIgnore:                 false
    HintName:                   
    HintPartitionable:          false
    HintSymbolicIconName:       
    HintSystem:                 true
    Id:                         by-id-dm-name-cryptswap
    IdLabel:                    
    IdType:                     swap
    IdUUID:                     014c237b-eae8-425f-a254-18ae81b8d8b5
    IdUsage:                    other
    IdVersion:                  1
    MDRaid:                     '/'
    MDRaidMember:               '/'
    PreferredDevice:            /dev/mapper/cryptswap
    ReadOnly:                   false
    Size:                       4294442496
    Symlinks:                   /dev/disk/by-id/dm-name-cryptswap
                                /dev/disk/by-id/dm-uuid-CRYPT-PLAIN-cryptswap
                                /dev/disk/by-uuid/014c237b-eae8-425f-a254-18ae81b8d8b5
                                /dev/mapper/cryptswap
    UserspaceMountOptions:      
  org.freedesktop.UDisks2.Swapspace:
    Active:             true

/org/freedesktop/UDisks2/block_devices/loop0:
  org.freedesktop.UDisks2.Block:
    Configuration:              []
    CryptoBackingDevice:        '/'
    Device:                     /dev/loop0
    DeviceNumber:               1792
    Drive:                      '/'
    HintAuto:                   false
    HintIconName:               
    HintIgnore:                 false
    HintName:                   
    HintPartitionable:          true
    HintSymbolicIconName:       
    HintSystem:                 true
[... 1811 lines cut ...]
    Running:                    true
    Size:                       456122171392
    SyncAction:                 
    SyncCompleted:              0.0
    SyncRate:                   0
    SyncRemainingTime:          0
    UUID:                       3687b83a:7842901d:e1816682:ab70c7ad

/org/freedesktop/UDisks2/mdraid/ec621a89_b2608e6b_9a4e3435_3fc10ba1:
  org.freedesktop.UDisks2.MDRaid:
    ActiveDevices:              [('/org/freedesktop/UDisks2/block_devices/sda', -1, ['spare', 'external_bbl'], 0, {}), ('/org/freedesktop/UDisks2/block_devices/sdb', -1, ['spare', 'external_bbl'], 0, {})]
    BitmapLocation:             
    ChildConfiguration:         []
    ChunkSize:                  0
    Degraded:                   0
    Level:                      container
    Name:                       
    NumDevices:                 2
    Running:                    true
    Size:                       0
    SyncAction:                 
    SyncCompleted:              0.0
    SyncRate:                   0
    SyncRemainingTime:          0
    UUID:                       ec621a89:b2608e6b:9a4e3435:3fc10ba1

mosler99, 2020-06-23:
no idea what it means but thats the dump.

also i get the same hang wehter writing to sd card or a file,

rgaudin, 2020-06-24:
OK, when you ran it in the terminal, did it ask for your password or open such a dialog?

mosler99, 2020-06-24:
i didn't run it from the terminal just double clicked it. when i attempt running it from the terminal it tells me "command not found" even when done as SU. 

I am relatively new to linux.

rgaudin, 2020-06-24:
Sorry I meant the udisksctl command. One possibility for the process to get stuck like this is that your user can't directly call udisksctl dump and it is waiting for you to answer a prompt (that you don't see).

So that's why I asked you to open a terminal and input `/usr/bin/udisksctl dump`. When you do this, do you get a prompt on the terminal asking for your password, or even a dialog?

mosler99, 2020-06-24:
ahhhh my bad on that one. no no password requested
dump is as follows. just in case...

 org.freedesktop.UDisks2.Manager:
    DefaultEncryptionType:      luks1
    SupportedEncryptionTypes:   luks1
                                luks2
    SupportedFilesystems:       ext2
                                ext3
                                ext4
                                vfat
                                ntfs
                                exfat
                                xfs
                                reiserfs
                                nilfs2
                                btrfs
                                minix
                                udf
                                f2fs
                                swap
    Version:                    2.8.4

kelson42, 2020-07-07:
@rgaudin What are the next steps here?

rgaudin, 2020-07-07:
> @rgaudin What are the next steps here?

Installing PopOS in a VM and try to replicate

rgaudin, 2020-07-07:
I'm sorry but I could not reproduce your issue with Pop! OS 20.04

First run complained about extfs and second passed that `udisksctl` step and directly after asked for my password.
[image]

kelson42, 2020-07-15:
I believe @rgaudin has finally identified the problem. `udisksctl` has a special behaviour around STDOUT/STDERR. This is written in the documentation: "This will deadlock when using stdout=PIPE or stderr=PIPE and the child process generates enough output to a pipe such that it blocks waiting for the OS pipe buffer to accept more data."

@mosler99 I suspect you have quite a few disks connected on your computer?

We need to deal with the output slightly differently and the problem should be fixed.

PR #524 (2020-08-13): subprocess_pretty_call to not use subprocess.wait
Fixes #513

According to doc, subprocess.wait() is not safe under some circumstances ; when the
ouput is large.
This prevent usdisks dump from working for some users on Linux

This uses subprocess.communicate() instead which has the disadvantage of not returning
all output line one by one, but stdout and stderr aside instead.
We're thus return STDERR after stdout.

Also, content is always decoded as the non-decoded version was not used anywhere.
stdin param and handling removed for same reason as well.

Ref: https://docs.python.org/3/library/su [... cut]
commit bf1f2e27 (2020-08-13): subprocess_pretty_call to not use subprocess.wait (#524)

* Fixed #513: subprocess_pretty_call to not use subprocess.wait

According to doc, subprocess.wait() is not safe under some circumstances ; when the
ouput is large.
This prevent usdisks dump from working for some users on Linux

This uses subprocess.communicate() instead which has the disadvantage of not returning
all output line one by one, but stdout and stderr aside instead.
We're thus return STDERR after stdout.

Also, content is always decoded as the non-decoded version was not used anywhere.
stdin param and handling removed for sa [... cut]
closed 2020-08-13 by rgaudin