offspot/kiwix-hotspot #514: Add clear link/help to open a bug

kelson42, 2020-06-27:
For the moment there is no help telling the users where to report a bug or request a new feature.

As a consequence people use various solution to be in touch with us (forum, email, irc, ...) and most of the time to go the the Github bug tracker. This is not exactly what we want.

We should be able to better help users to find the Kiwix Hotspot bug tracker. A few idea:
* Create a new menu item pointing directly to the Github tracker
* Create a new paragraph in the FAQ/Help wiki page
* Create a link from Kiwix-Hotspot (admin?) web page itself.

Popolechien, 2020-06-27:
I've added a button on the website's contact form that point to /issues. We could probably do with a blog post explaining how to post a _good_ issue ticket.

kelson42, 2020-06-27:
The how is clearly document https://github.com/openzim/overview/blob/master/REPORT_BUG.md. You can make a blog post if you want, but the problem is more the "where?" here.

Popolechien, 2020-06-27:
Yeah but aren't hotspot users offline? Having a link that points to nothing actionable would be poor UX imho.

kelson42, 2020-06-27:
Yes, this is why my first proposal is "menu item" in the Kiwix Hotspot Installer.