offspot/kiwix-hotspot #515: The requested URL "/[file]" was not found on this server

DBemis, 2020-07-02:
Kiwix Hotspot Installer Version: 2.3.2
Config: [256gbConfig.txt](https://github.com/kiwix/kiwix-hotspot/files/4867045/256gbConfig.txt) (note that I've also tried with just a few resources selected and had the same issue)

MicroSD: [Samsung Evo Select 256gb](https://www.amazon.com/gp/product/B0887P21Z2)
Device: Raspberry Pi Zero W v1.1
Build info: Built with the Hotspot Installer 2.3.2 x64, on Windows 10 1909, writing directly to the SD Card (I have also tried saving to an image directly, which produced the same result). I've left the settings as default, other than selecting various static content

Issue: 
When attempting to go to any of the items on the Pi via the hot spot, I always get the error "Not found - The requested URL "/[filename]" was not found on this server."
[image]
[image]

Please let me know if you need any additional info - thanks!

rgaudin, 2020-07-03:
Can you please insert the SD card on your computer and look at the content of the *data* partition?
Your ZIM files should be there. Try to open one from there and check that it's working.

kf6nvr, 2020-07-03:
I have this same problem. Brand new installation from the last couple of hours (as of this post). 

While you asked someone else, here's my answer for the `HSPOT_DATA` partition that you're referring to:

Under `packages` the 37.3GB Wikiepedia file named `wikipedia_en_all_nopic.en.zim` exists. The browser URL to it ends in `.en`.

That is, when choosing to go to the Wikipedia content from the home screen `http://kiwix.hotspot` it then tries to load `http://kiwix.kiwix.hotspot/wikipedia_en_all_nopic.en` but nothing is there. 

My setup is on a Pi 4, though given everything else is working and the error is the same as on the Pi Zero W above, that's probably irrelevant. It seems like a web server configuration error.

DBemis, 2020-07-04:
> Can you please insert the SD card on your computer and look at the content of the _data_ partition?
> Your ZIM files should be there. Try to open one from there and check that it's working.
[image]
I loaded up a zim file from the SD without issue.

Another data point - I've tried smaller sets (but with still a good amount selected) previously with the same error (sadly I didn't save the config for those). However, today I grabbed a very small set of static content, and they did work - [samplecreation.txt](https://github.com/kiwix/kiwix-hotspot/files/4871909/samplecreation.txt)

I'm not sure what that means - too many selections causing issues? Maybe one specifically causing all to fail? Or a conflict when some combo of them are selected?

kelson42, 2020-07-04:
@DBemis @kf6nvr Thank you very much for your reports. We have remarked that recent ZIM files in combinaition with older `kiwix-serve` (the software in charge of serving the ZIM content) lead to an abnormal memory consumption. The problem has been fixed and a new kiwix-serve has been released. We should bring it in a new Kiwix-Hotspot release soon. Hopefully this would fix the problem you report. Really sorry for the inconvenience.

DBemis, 2020-07-04:
> @DBemis @kf6nvr Thank you very much for your reports. We have remarked that recent ZIM files in combinaition with older `kiwix-serve` (the software in charge of serving the ZIM content) lead to an [...]

Awesome! Thanks for the update 👍

rgaudin, 2020-07-06:
Please try with version [2.3.4+](http://download.kiwix.org/release/kiwix-hotspot/) and (re)open a ticket if you still encounter any issue.

closed 2020-07-06 by rgaudin