offspot/kiwix-hotspot #516: The requested URL "/[file]" was not found on this server

DBemis, 2020-07-08:
Kiwix Hotspot Installer Version: 2.3.4
[image]

Config: [New256gbConfig.txt](https://github.com/kiwix/kiwix-hotspot/files/4887826/New256gbConfig.txt)

MicroSD: Samsung Evo Select 256gb
Device: Raspberry Pi Zero W v1.1
Build info: Built with the Hotspot Installer 2.3.2 x64, on Windows 10 1909, writing directly to the SD Card (I have also tried saving to an image directly, which produced the same result). I've left the settings as default, other than selecting various static content

Issue:
When attempting to go to any of the items on the Pi via the hot spot, I always get the error "Not found - The requested URL "/[filename]" was not found on this server." This was thought to have been fixed with [issue 515](https://github.com/kiwix/kiwix-hotspot/issues/515), but I am getting the error after building a new image
[image]
and
[image]

Sample of images on SD Card:
[image]

Desktop app testing from files on SD Card:
Tried to load wikipedia_en_all_maxi.en.zim:
[image]

Tried to load ted_en_science.en.zim and got the same as above

Successfully loaded crashcourse_en_all.en.zim
[image]

Successfully loaded ted_en_global_issues.en.zim
[image]

Let me know if you need additional info
Thank you!

rgaudin, 2020-07-08:
Ah. my bad, sorry. I know what this is. Will fix

kelson42, 2020-07-08:
Arghhhh

rgaudin, 2020-07-08:
Really sorry about that and the time you wasted on this. The fix wasn't incorporated in 2.3.4 and thus it could not have worked.
It now is in [2.3.5](http://download.kiwix.org/release/kiwix-hotspot/) (kiwix-serve is updated at configure stage). Please kindly retry with that new version.

closed 2020-07-08 by rgaudin