offspot/kiwix-hotspot #517: Installation Failed error

DBemis, 2020-07-09:
Sorry - back with another

Kiwix Hotspot Installer Version: 2.3.5
Config: [SimpleInstall.txt](https://github.com/kiwix/kiwix-hotspot/files/4893969/SimpleInstall.txt)

Issue: During install, I get the error:
[image]

Log file: [kiwix failed install log.txt](https://github.com/kiwix/kiwix-hotspot/files/4893973/kiwix.failed.install.log.txt)

There are a couple errors before it gives up that I noticed, in case it helps:
`/usr/local/bin/kiwix-serve: unrecognized option '--version'`
And:
`"msg": "Request failed: <urlopen error timed out>", 
"state": "absent", 
"url": "http://download.kiwix.org/release/kiwix-tools/kiwix-tools_linux-armhf-3.1.2.tar.gz"`

I verified I can directly download kiwix-tools_linux-armhf-3.1.2.tar.gz

Let me know if you want me to test anything

rgaudin, 2020-07-09:
Thank you for the report.

The `unrecognized option '--version'` is normal ; that's expected as bundled version didn't have that option. It correctly triggered the download mechanism (that's your fix for the other ticket).

Thank you for trying several times . I will launch a new test myself and come back to you. Will probably send you a test build once I identify the problem. Are you on Windows?

DBemis, 2020-07-09:
> Thank you for trying several times . I will launch a new test myself and come back to you. Will probably send you a test build once I identify the problem. Are you on Windows?

Thanks! Yes - Windows 10 x64 1909, 18363.900

rgaudin, 2020-07-11:
Thank you for your patience, please test the [following build](http://tmp.kiwix.org/ci/kiwix-hotspot-win64-kiwixserve.exe).
If you confirm this works (both installation and ZIM reading inside the hotspot), we'll make another release.

DBemis, 2020-07-11:
> Thank you for your patience, please test the [following build](http://tmp.kiwix.org/ci/kiwix-hotspot-win64-kiwixserve.exe).
> If you confirm this works (both installation and ZIM reading inside the hotspot), we'll make another release.
Thanks @rgaudin! 

I completed a test with the config file from the original post (which just has a single, smaller zim), and it completed successfully! Running the larger test now - probably won't have results of that until tomorrow since it takes quite a while to run

Thanks again!

DBemis, 2020-07-12:
@rgaudin - tested out the full install (~250gb worth), and everything I tested works perfectly! I went into 20+ of them, plus all of the larger ones I had, and everything works. Thanks again for the digging and figuring it out 🥳

rgaudin, 2020-07-12:
Very glad to read it. Thank you again for the reports !

commit d20b8a9b (2020-07-12): Fixed #517: network operations not working inside qemu
closed 2020-07-12 by rgaudin