offspot/kiwix-hotspot #520: Smarter download checksum computation

kelson42, 2020-07-16:
I'm building a 200GB Kiwix-Hotspot image and my computer takes more than one our to (re-)check checksums in my cache although this already have been yesterday on the very same files. This seems to me unecessary and we could greatly speed-up the whole process by avoiding redundant checks.

kelson42, 2020-07-16:
I see two topics here:
* It is important to check that downloaded files are proper but I believe aria2c does that already by using the metalink? So why having a custon check?
* We need to secure once that the local cache content is the same as online. This shoudl be done at the download time. At the cache re-using time, we should not do that check again, as this has been secured already.

rgaudin, 2020-07-16:
Actually that's already what's being done

https://github.com/kiwix/kiwix-hotspot/blob/de6d824a8cafa11557a307adc9b1273b1a1ec9ff/kiwix-hotspot/backend/download.py#L223-L236

If file exist, we verify if the sum matches what's expected (that's where you spent time on today) and download if not.
If the file doesn't exist, we download it and don't verify checksum.

kelson42, 2020-07-16:
@rgaudin OK, then we even have an other question on the top of the two I have already written:
* We should secure after the download that the file are OK? Or do we consider this secured by `aria2c` OK (see my first question)?

rgaudin, 2020-07-16:
We don't supply the checksum manually to aria2 so at the moment it only works for metalink URLs.
All our content are under http://download.kiwix.org so we have metalink.

The only other file we download is raspbian base image (at master creation step) and that is not metalink so we just *assume* that if aria2c is happy (without checksum) then it's OK.

kelson42, 2020-07-16:
It might be better to copy the raspbian image to download.kiwix.org, to simplify the problem.

For the rest it seems, indeed, that we can not trust `aria2c` to check the checksum considering how we use it, but if we would use the the following options, it would I believe make any other test useless:
```
       -V, --check-integrity [true|false]
              Check  file  integrity by validating piece hashes or a hash of entire file.  This option has effect only in BitTorrent, Metalink downloads with checksums or HTTP(S)/FTP downloads with --checksum option.  If piece hashes are provided, this option can de‐
              tect damaged portions of a file and re-download them.  If a hash of entire file is provided, hash check is only done when file has been already download. This is determined by file length. If hash check fails, file is  re-downloaded  from  scratch.   If
              both piece hashes and a hash of entire file are provided, only piece hashes are used. Default: false
```

and

```
       --checksum=<TYPE>=<DIGEST>
              Set  checksum.  TYPE  is  hash  type. The supported hash type is listed in Hash Algorithms in aria2c -v. DIGEST is hex digest.  For example, setting sha-1 digest looks like this: sha-1=0192ba11326fe2298c8cb4de616f4d4140213838 This option applies only to
              HTTP(S)/FTP downloads.
```

But that could be straight, considering that the checksum is in the yaml. Right?

kelson42, 2020-07-16:
Or use directly metalink links directly, we would have then a dependence to #366

rgaudin, 2020-07-16:
No, aria2 automatically checks the file because the chunk checksums are provided in the metalink. That's why we don't additionally supply them.

And you're right, for non-metalink, we could pass the checksum as we have it at hand.

rgaudin, 2020-07-16:
> Or use directly metalink links directly, we would have then a dependence to #366

I don't understand that. Nor the linked issue :D

kelson42, 2020-07-16:
@rgaudin Sorry, I meant #170

rgaudin, 2020-07-16:
What is the checksum problem to solve exactly ?

kelson42, 2020-07-16:
@rgaudin You need to retrieve it (from the Yaml, from the OPDS), then you have to share it to aria2c via `--checksum`. This is more complex and error prone than just giving a metalink link to download.

rgaudin, 2020-07-16:
I think we're not on the same page:

* aria2c automatically gets metalink from a link on our server. See bellow.
* aria2c automatically validates checksum from metalink if the chunk checksums are provided (even without --checksum). Which is the case for us.

```
curl -H "Accept: application/metalink4+xml" download.kiwix.org/zim/other/mali-pour-les-nuls_fr_all_2020-07.zim
```

kelson42, 2020-07-16:
@rgaudin OK, then why not just deactivating the custom checksum? At least for anything implying download.kiwix.org?

rgaudin, 2020-07-16:
The custom checksum is only used when retrieving from cache without downloading. It's there to ensure the file is same as remote. It could have been downloaded the day before (as for your case) or 6 months before or be altered or be a completely different file with same name. Given the time of the overall process and that most of those files are used at the very end of the process, it makes sense to make sure early that those are OK.

But I agree that for users with super fast connexion, it takes a lot longer to calculate hash than to re-download super large files. You could wipe your cache to avoid that

kelson42, 2020-07-16:
@rgaudin This seems an extremly high price to pay for the benefit. Like we have seens, I'm not even sure that all the mirroring system works fine by handling the case were we re-publish a new version on the very same filename. But if this has to be done - what I really put in question - then why not just checking the filesize? or just using `aria2c` with the `---continue` option?

rgaudin, 2020-07-17:
I agree with the fact that we could be less strict and that would be OK for most cases but having this is a security belt and I strongly disagree with this being “*an extremly high price to pay*”.

On my old laptop, it takes 3mn30s to calculate the sum of this kalite_videos_en.tar file (46GB). So linearly expanding that to a 256GB full card that would be less than 18mn. That seems reasonable to me.

kelson42, 2021-12-03:
We can not really do better, unfortunately. The checks which have to be done are already done.

closed 2021-12-03 by kelson42