offspot/kiwix-hotspot #526: Add 10% margin to SD card space before allowing card creation

Popolechien, 2020-08-27:
Someone created a 228.88 Gb (245.76 GiB) configuration. System went for a 256Gb card but crashed shortly after the order was started. I reused the same configuration but removed some content to go down to 214.83 Gb (230.67GiB). It worked fine, so I would recommend forcing some extra unused space (10%?) before launching an order.

I've created an order at 224.55 GiB/241Gb that seems to be going ok as well, so 5-6% could also be an acceptable upper bound.

rgaudin, 2020-08-27:
If I recall correctly the margin is not linear (10% of 250G is 25G!!).

I'll look into it as this is kind of a try and fix game.

rgaudin, 2021-04-21:
OK, I think we were mislead by the proposal to add margin inside the card. That's not what the problem is about.

The problem is that a user asks for an image of size `x` with a list of content resulting in size `y` and `x < y`. The installer fails because that's what it should do and that's why I'm closing this ticket as invalid.

The root cause is in the cardshop.

We have a set of calculations for accounting all packages and matching with SD-card sizes with margins and stuff. We are using a duplication of the same code for those both in cardshop and hotspot.
Were we to have edits in this code in one or the other without replicating, it could lead to differences ending up in failures like this, but this did not happen because we've been careful about that.

What happens I believe is that user create a Configuration with various content and its size is computed on save.
When user order this configuration, we just use the size from that config that might have been created months ago.
When the installer starts it, it computes the actual size with the list of content, which refer *Book* and not specific ZIM. So if `wikipedia_fr_all_maxi` was on a config and there was not much more space left, whenever the ZIM is refreshed, chances are the size of that ZIM increased significantly. So sum of ZIM sizes exceeds the in-config value and potentially the image requested size.

What we should do: Recompute sizes on Order creation and stop it right away if the size if not valid anymore. Very similar to the missing zim problem. Opening a ticket now.

Popolechien, 2021-04-22:
@rgaudin
> What happens I believe is that user create a Configuration with various content and its size is computed on save.
When user order this configuration, we just use the size from that config that might have been created months ago.

This is not what happened in these two cases https://cardshop.hotspot.kiwix.org/all-orders/607d55d698c8bf0023256759 and https://cardshop.hotspot.kiwix.org/all-orders/607d5f9c98c8bf0023256918 : the configuration were created mere _minutes_ before the order so it's unlikely that zim file size has changed substantially during that short timespan.

rgaudin, 2021-04-22:
This is not a hotspot issue. It's a cardshop one. As I explained above, the cardshop calculates appropriate Card size for a list of content. That's what's wrong here. If the installer refuses it, the cardshop should not accept it and send it to the installer.

Now the reasons for this are three fold:
- different calculation code (hotspot doesn't match cards for content but only validates a requested content and sizes are compatible)
- library off-syncing (cardshop one vs online)
- content size change since configuration

closed 2021-04-22 by rgaudin