offspot/kiwix-hotspot #527: Add LIVE (real-time) collaboration tools like Etherpad and Jitsi servers

zblace, 2020-09-01:
Hi! 
It would be awesome if kiwix-hotspot could include optional LIVE (real-time) collaboration tools like Etherpad and Jitsi servers that could be used to do and demo workflow of education/ collaborations in physical proximity (with pandemic minimum distance). Anyone interested to work on this?

Best Z

rgaudin, 2020-09-01:
Sounds like a good idea but I'm not sure how capable is a raspi for [jitsi](https://jitsi.github.io/handbook/docs/devops-guide/devops-guide-scalable#machine-sizing)

Popolechien, 2020-09-01:
Yeah I'm not sure I'd see the use case (unless on a mesh network but then that's a long way away.

Etherpad on the other hand could easily come in handy.

zblace, 2020-09-21:
> Sounds like a good idea but I'm not sure how capable is a raspi for [jitsi](https://jitsi.github.io/handbook/docs/devops-guide/devops-guide-scalable#machine-sizing)

My wild guess that version 4 would be able to handle 10 local area requests 
*(enough for doing offline workshops with distance for pandemic)...
...maybe also older versions if it is headless system?

zblace, 2020-09-21:
> Yeah I'm not sure I'd see the use case (unless on a mesh network but then that's a long way away.

@Popolechien 
Even if serving just on LAN for 10 people that would be enough for doing offline workshops in pandemics *(keeping the distance, but demonstrating services and tools of online work)...
But I am interested why do you say mesh network is long away?

Popolechien, 2020-09-21:
@zblace simply because it's not on our tech road map. But there are mesh-based projects running Kiwix already here and there.

kelson42, 2021-12-04:
Similar to #453?

Popolechien, 2021-12-04:
@kelson42 I do not think so - my understanding of Beekee is that it is a suite of education tools (incl. probably Moodle), but not sure Jitsi/Etherpad are on there.