offspot/kiwix-hotspot #528: Nginx fails to start on boot, but 'sudo systemctl restart ngnix-service' gets it going

jacycatlin, 2020-09-17:
Using kiwix-hotspot 2.3.6 on Mint 20 to make an image on a 128Gb Sandisk Ultra for a Raspberry Pi Zero W -

As visible in the attached pics, there is an error or two during boot (pic 1)
Once you log in (can't use the username/pwd I put in the GUI, have to use pi default login) you see the details (pic 2)
Simply restarting the service - no other fixes or adjustments - brings it online as you'd normally expect.

[image]
[image]
[image]

Apologies for pics instead of text but I'm in a hurry... thanks for the great software though!

kelson42, 2020-09-17:
@jacycatlin Thank you for your bug report, Please give a us a few days to give you a proper feedback.

rgaudin, 2021-04-15:
Thank you for your report and sorry for the delay. Please test again with upcoming version 2.4 once it is out (early next week). You'll notice the service has been renamed `kiwix` meanwhile.

[image]

closed 2021-04-15 by rgaudin