offspot/kiwix-hotspot #529: macOS installer probably needs a certificate update

Popolechien, 2020-09-17:
I got the below message and am not allowed to use the application:
[image]

rgaudin, 2020-09-17:
Duplicate of #497. You can bypass this, as for other un notarized software by right-clicking the app and selecting open.

closed 2020-09-17 by rgaudin

Popolechien, 2020-09-17:
Nope, cannot be bypassed anymore (lest I need to open up some other authorizations in my preferences maybe?)

rgaudin, 2020-09-17:
Just tested and works as I described.
[image]
[image]
[image]
[image]

Popolechien, 2020-09-17:
Ok, works now for me as well. No idea what I missed or messed. Thanks.