offspot/kiwix-hotspot #530: Update macOS installer logo

Popolechien, 2020-09-17:
The Installer is still using the old logo
[image]
We could / should probably replace it with the logo that was kindly designed a while back by a user:
[image]
(this is a .png but I can provide an .svg if need be)

**Question: ** is it mandatory that both logos (top and bottom) be identical?

rgaudin, 2020-09-17:
What is *the old logo*? That's the only one I know.
I guess you're referring to the *Kiwix* logo that serves as a base for this one. Then we could do a new version based on the new kiwix logo.

The contributed logo your attached has been made for the cardshop (and I think it corresponds better).

To answer the question, no, both can be different. Top one is just a branding we set and bottom one is the app logo.

Share a new logo for the app and/or the dmg background if you want it updated.

Popolechien, 2020-09-17:
Well both (cardshop and installer) do pretty much the same thing, so it does not strike me as irrelevant. I'd be more wary of logo variants going too far off one another, but in the case at hand we had a ready-made solution to use, so why not.
I'll ping @siemsie and see what they can do.

siemsie, 2020-09-23:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
Hello tout le monde – voici quelques premières idées. Feedback? Merci :)

rgaudin, 2020-09-23:
Merci ! Je les trouve tous très bien. Je préfère quand il y a un peu de couleur vs monochrome seul mais c'est tout.

Popolechien, 2020-09-23:
Alors plutot le numéro 2, et tant qu'à faire créer un logo d'app qui soit identique / similaire / homogène avec celui qu'on a fait pour macOS dans [230](../../apple/issues/230)

rgaudin, 2020-09-23:
Attention, ce que tu as partagé c'est pour l'appli macOS/iOS et c'est adapté à ces plateformes. Ici tu as ouvert le ticket avec le screenshot macOS mais on utilise la même icone pour les trois plateformes.
Donc faut vérifier si ça vous va aussi pour les autres plateformes, ou faire des adaptations (on est pas obligé d'avoir la même icone).

Popolechien, 2020-10-05:
@DhairyaBahl please use logo #2 above but check if it fits on all platforms in terms of size.
@siemsie Can you please create a macOS-specific round button like there is for the reader? 

[image]

DhairyaBahl, 2020-10-05:
@Popolechien Okay I got it. I am working on the issue. Thanks

siemsie, 2020-10-07:
@Popolechien Can you check if this doc is ok? I based myself on what I handed in last time... 
[Kiwix-icon-MacOS-Installer.zip](https://github.com/kiwix/kiwix-hotspot/files/5342108/Kiwix-icon-MacOS-Installer.zip)

Popolechien, 2020-10-08:
@siemsie looks good, thanks!

DhairyaBahl, 2020-10-08:
@Popolechien Okay I got. Working on it. :)

rgaudin, 2021-04-20:
Done in #569

closed 2021-04-20 by rgaudin