offspot/kiwix-hotspot #531: Kiwix Hotspot installer is not working on Win 10

Pospitch, 2020-09-17:
I downloaded Kiwix Hotspot installer kiwix-hotspot-win64.exe and tried to install it. It extracted files and after that I got fatal error: Failed to execute script __main__

My system version:
Windows 10 Home
Version 1909
Build 18363.1082

rgaudin, 2020-09-17:
hotspot version ? 2.3.6 release ?

rgaudin, 2020-09-18:
[image]

Tested ; works with 2.3.6. Please confirm which version you were using and reopen if it's 2.3.6.

Pospitch, 2020-09-18:
I cannot check the version as the aplication won't run. I downloaded the latest version from oficial website. If there is a other place to download, where is written version, let me know please.

[VirtualBox_w10_18_09_2020_09_10_22]<https://user-images.githubusercontent.com/57929/93580139-e7ed9200-f98e-11ea-8c77-e9ce91fae43f.png>

rgaudin, 2020-09-18:
http://download.kiwix.org/release/kiwix-hotspot/v2.3.6/ is the only location I know. You wrote you installed 64bit version, are you sure the Windows is 64b as well (right-click on Computer gives you that information)

rgaudin, 2020-09-18:
btw the link in kiwix.org website is OK

```
 curl -I -L http://download.kiwix.org/release/kiwix-hotspot/kiwix-hotspot-win64.exe
HTTP/1.1 301 Moved Permanently
Server: nginx/1.17.6
Date: Fri, 18 Sep 2020 09:19:20 GMT
Content-Type: text/html; charset=iso-8859-1
Location: http://download.kiwix.org/release/kiwix-hotspot/v2.3.6/kiwix-hotspot-win64.exe
```

Pospitch, 2020-09-18:
Of course I have 64bit Windows, I'm not even sure if there is such thing as
32bit Windows 10. I tried to download it again. And same problem. I run
that file, it extracted files and right after extraction I got same error.

rgaudin, 2020-09-18:
OK, will try to reproduce on a fresh W10. If it's easy for you, a screenshot would help.

Pospitch, 2020-09-18:
Is this enough? I hope I can send them as email attachments. Also, I
managed to get another laptop, so I will test it there too.

Pospitch, 2020-09-18:
So I tried on another 64bit Windows 10 laptop and got exactly same error.

Popolechien, 2020-09-18:
FWIW I just downloaded and tested on my own machine (W10 version 1903, build 18362.1016) and it works fine. There's an update coming up (presumably to OP's version), I'll see if it affects anything in anyway.

rgaudin, 2020-09-21:
Just installed a fresh VM off the microsoft-distributed ISO and run into no issue.
Closing this until we get something reproducible/actionable.
[image]

closed 2020-09-21 by rgaudin

Pospitch, 2020-10-08:
Hello,

I just wanted to say, that I tried it on 4 different computers and I have
same problem on all of them. So it seems that problem is really in that
application.