offspot/kiwix-hotspot #532: image file should be name after configuration rather than scheduler ID

Popolechien, 2020-09-18:
Scheduler ID is unique, but the odds of remembering what's on an image based on that seemingly random number are quite low.
I could see the issue of users using the same config name and near-simultaneously asking for a card creation, but in that case a discreet `confignamex` (with `x`= queue ID or something similar) could do the trick.

rgaudin, 2020-09-21:
Couple of points:
- why would one want to remember what's on an image file?
- I think creating multiple images around the same period (differentiating 2 images) is not a typical use case, except for you.
- we need unique filenames, both on the filesystem and on the URL (uses filename)

That said, I agree current filenames are not practical: if not renamed right after download, only size would give a hint about what that file is.
I suggest we go beyond what you propose and use something like `cardshop-<some-id>-<config-name>_2020-09-21.img` for instance. `cardshop` would be to remember what that big file (with no metadata on most OS) is about, the `id` for uniqueness.
Now you tell me whether config-name and date are actually useful or not. Date could be retrieved from file metadata, most of the time.

Popolechien, 2020-09-28:
> why would one want to remember what's on an image file?

People who manage several locations for instance.

> we need unique filenames, both on the filesystem and on the URL (uses filename)

Yes indeed but the file are exported / downloaded can be renamed when stored on the warehouse. If a similar config is created then the pratice is to append (n) at the end of the file (e.g., `configname.zim` and `configname_(1).zim`)

kelson42, 2021-04-20:
@Popolechien @rgaudin Two remarks:
* Should we not fix it like https://github.com/openzim/zimit-frontend/issues/13
* It is not a ticket for the cardshop?