offspot/kiwix-hotspot #533: Issue during install

VerstreuteSeele, 2020-09-20:
Installed as described on your page. Using a good quality 32GB micro-sd Card and newest version of ubuntu

Failed: Process ['/sbin/mkfs.exfat', '-n', 'HSPOT_DATA', '
Installation failed: Process ['/sbin/mkfs.exfat', '-n', 'HSPOT_DATA', '/dev/loop15'] failed

--- Exception Trace ---
Traceback (most recent call last):
  File "run_installation.py", line 329, in run_installation
  File "backend/mount.py", line 396, in format_data_partition
  File "backend/util.py", line 110, in subprocess_pretty_check_call
  File "backend/util.py", line 102, in subprocess_pretty_call
backend.util.CheckCallException: Process ['/sbin/mkfs.exfat', '-n', 'HSPOT_DATA', '/dev/loop15'] failed

VerstreuteSeele, 2020-09-21:
little update: Worked perfectly fine on windows and I now have a new usecase for my old raspberry pi. 

But i'd still love to figure out why it didnt work on Ubuntu.

rgaudin, 2020-09-21:
I don't know what went wrong. Here are a few manual steps you could try to input in a terminal so we get a feedback of mkfs.
Please share its output.

```sh
# inside the kiwix-hotspot.cache folder
unzip hotspot-master_2019-08-24.img.zip
 # get the loop device path for data partition : /dev/loopX (once you get it, replace X with number in following commands)
/usr/bin/udisksctl loop-setup --offset 7000031744 --size 755805696 --file hotspot-master_2019-08-24.img --no-user-interaction
# make sure not mounted
/usr/bin/udisksctl unmount --block-device /dev/loopX --no-user-interaction
# change perm
sudo chmod -v o+rw /dev/loopX
# format loop
mkfs_exe -n HSPOT_DATA /dev/loopX
echo $?
# now remove what we did
/usr/bin/udisksctl unmount --block-device" /dev/loopX --no-user-interaction
/usr/bin/udisksctl loop-delete --block-device /dev/loopX --no-user-interaction
rm hotspot-master_2019-08-24.img
```

VerstreuteSeele, 2020-09-21:
Looks like I found the issue:
Soo.. if I run "mkfs_exe -n HSPOT_DATA /dev/loopX" I get: command not found. 
If however I run just mkfs instead of mkfs_exe I get this error: "mkfs.ext2: invalid blocks '/dev/loop11' on device 'HSPOT_DATA' "

rgaudin, 2020-09-21:
Sorry it's a typo it should be mkfs.exfat instead of mkfs_exe.
I hope you replaced the X in /dev/loopX in all the commands as well

kelson42, 2021-01-14:
@rgaudin I fear that without any feedback from @VerstreuteSeele it might be difficult to do anything here?!

rgaudin, 2021-01-14:
Yes, let's close it.

closed 2021-01-14 by rgaudin