offspot/kiwix-hotspot #535: Increase maximum image size to 512 Gb

Popolechien, 2020-10-02:
The current maximum size that can be generated on the Cardshop is 256 Gb, but we've had instances of people maxing this out and, with the increase in the amount of videos we offer, this size does not seem this much far fetched anymore (and someone trying to generate a StackExchange + enwp image would also be very close to the maximum size allowed already. 

400Gb and 512Gb cards are now commercially available, it therefore only makes sense for us to follow the market and be able to provide images within this size range.

DhairyaBahl, 2020-10-05:
I would like to work on this issue.

rgaudin, 2020-10-05:
This is a cardshop-only issue where we list options. AFAIK, there is no limitation on the hotspot installer.

closed 2020-10-05 by rgaudin