offspot/kiwix-hotspot #540: Include Kiwix APK as downloadable app

Popolechien, 2020-10-13:
Just as we already offer Math Matthews and other apps it would be useful to be able to download the Kiwix app from the hotspot.

Use case: Hotspots are being used in refugee camps. Users get trainings from the MOOCs installed on the hotspots, but the idea is that they should keep using the MOOC elements even as they continue their journey. Hence, they could download the Kiwix APK and from there the various zim files they need, eventhough at no point they were connected to the internet.

Related issue: [#411](../../kiwix-tools/issues/411)

rgaudin, 2020-10-13:
👍  but without a way to download ZIM files from the hotspot, it's essentially useless.

I don't think your 411 issue will lead to something. kiwix-serve already mimics download.kiwix.org as… it's what powers it.
What's missing is:
- a web server to expose the zim file (hotspot config)
- kiwix serve to be told and to propagate where the ZIM files are (there's a ticket somewhere about that)
- spoofing download.kiwix.org… that'd be hotspot config but if the URL in the app is https, it won't work.
I wonder if it would be wiser to have something explicit in the app to switch to a local library if it exists rather than faking the online thing (leading to uncertainty and issues). It would allow the same re-download from any kiwix-desktop on the network

Popolechien, 2020-10-13:
> a web server to expose the zim file (hotspot config)

So should I close 411 on Kiwix-tools and re-open a similar one here?

kelson42, 2020-10-29:
@rgaudin For now it downloads from `http://mirror.download.kiwix.org/`.

kelson42, 2020-10-30:
Duplicate of #366

closed 2021-04-26 by rgaudin

rgaudin, 2021-04-26:
@kelson42 it's #365