offspot/kiwix-hotspot #542: Images do not run on Pi3B+

Popolechien, 2020-11-05:
I created two distinct cards that I tried on two Raspberry Pi 3B+ : the network would not appear. I tried both on a W (zero) and they both worked fine.

rgaudin, 2020-11-05:
Can you please test your Pi3B+ with another card?

Popolechien, 2020-11-05:
@rgaudin You mean a third configuration? Sure.

rgaudin, 2020-11-05:
Not a new configuration ; I mean another card. Goal is to test your Pi is working

Popolechien, 2020-11-05:
I have tried on a third device and the network did not appear on my Mac, but fleetingly did on my android phone and then disappeared (never to be seen again).

Popolechien, 2020-11-05:
Just to clarify:
Card 1 (128Gb config A): works on W; does not work on R3B+ n°1, n°2 nor n°3
Card 2 (256Gb config B): works on W; does not work on R3B+ n°1, n°2 nor n°3

rgaudin, 2020-11-05:
Have you used any of those two cards before? What's their size?
Given nothing has changed, I lean toward a card incompatibility so if you could use a previously-working or smaller card with a Pi 1, 2 or 3, that should confirm this hypothesis.

Could you share the brand/model of those new cards as well?

Popolechien, 2020-11-05:
Card 1: Sandisk 128Gb Extreme Pro U3/A2 - has been used before
Card 2: Samsung 256Gb Evo Plus Class10 / U3 - first use

I do not have a Pi 1, 2 or 3 but will try with another, smaller card.

rgaudin, 2020-11-05:
How can you say it doesn't work on R3B+ n°1, n°2 nor n°3 if you don't have those devices?

Popolechien, 2020-11-05:
Here is what I have:
- Three RPi 3B+
- One RPi W
- One 256Gb and one 128Gb microSD card

What I do not have (and probable source of misunderstanding):
- Older Raspberry Pi model 1, 2 or 3, or even the newer RPi 4
[image]

rgaudin, 2020-11-06:
Tested the demo image on a 3B+ without issue.

[image]

I suggest you test the demo image on one of those cards and test. If it works, it's a software issue that we'll have to look into. If it doesn't, it's most probably a card-pi compatibility issue that we can't do much about.

Popolechien, 2020-11-13:
Ok so here is where I stand now with my tests (Y= Network appears and can be connected to / N= did not work)

| Card/config | RPi W | RPi 3B+ `#1` | RPi 3B+ `#2` | RPi 3B+ `#3` |
| -- | -- | -- | -- | -- | 
| SanDisk 16 Gb |Y |N |N |N |N |
|SanDisk Ultra 32 Gb | Y |N |N |N |N |
|SanDisk Extreme Pro 128 GB | Y |N |N |N |N |
|Samsung 256Gb Evo Plus | Y |N |N |N |N |

We could probably argue that there is a compatibility issue with SanDisk and RPi 3B+, but that sounds pretty unlikely to me. I have nevertheless ordered a Pi4 and a non-SanDisk card, we will see.

rgaudin, 2020-11-13:
Thank you. It would be very helpful if you could plug the Pis to a monitor and let us know what is displayed (for the non-working scenarios). That would instantly clarify if its a card problem or if it's software/wifi related

Popolechien, 2020-12-01:
Ok, got my 4B and it works like a charm. Odd that 3 distinct 3B+ would not work though.

closed 2020-12-01 by Popolechien