offspot/kiwix-hotspot #544: Veriffy that we can connect enough clients to Kiwix-Hotspot

kelson42, 2020-11-11:
This ticket of IIAB is pretty worrying https://github.com/iiab/iiab/issues/823

Popolechien, 2020-11-11:
So, practically speaking, should we get X people together to try and connect to a device? (it's going to be difficult to pull together these days, tbh)

kelson42, 2020-11-12:
@tim-moody Do you know a way to simulate the wifi clients?

tim-moody, 2020-11-12:
@kelson42 my efforts are rather crude, but are at https://github.com/iiab/iiab-factory/tree/master/testing/wifi-testing

I was unable to establish multiple ip addresses on wifi devices, unlike on wired devices, so I used usb hubs with wifi dongles. This allowed me to have 10 wifi devices per raspberry pi for testing. I found 10 to be the point at which reliability was acceptable. More than that the usb hubs tend to disconnect after 30 minutes or so.

There is a small service that runs on the server to receive test requests and a python client that creates the connections and then makes requests and reports status.

This is the hardware I used:

https://www.amazon.ca/gp/product/B008IFXQFU/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1 

https://www.amazon.ca/gp/product/B014ZQ07NE/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1 

Plus rpi 4s . My latest config is I put one usb hub with 7 dongles and 2 additional dongles directly on the rpi4, which gives me 10 devices counting the builtin wifi.

kelson42, 2020-11-12:
@tim-moody Thank you very much for your quick and helpful answer. We will have a look, but this looks promising.

tim-moody, 2020-11-12:
@kelson42 hopefully you can publish your results.

Here is my summary of iiab/iiab#823 and some other findings.

https://iiab.zulipchat.com/#narrow/stream/231879-iiab-devel/topic/rpi.20wifi.20firmware

kelson42, 2020-11-12:
@tim-moody We are not even sure we are even impacted. Considering that this investigation takes a bit of time, we won't have a look to that before 2021. But this ticket will be then updated ASAP.

kelson42, 2021-07-25:
Here is an appliance doing that https://smartechconsulting.com/product/ixia-ixwlan-sed-wi-fi-device-and-network-tester/

kelson42, 2021-07-25:
Seems the Hawking HD45U supports up to 8 interfaces (https://www.cwnp.com/forums/posts?To-simulate-100-WLAN-clients-124128). With 4 of them (via USB switch), it should be possible to instrument them and test how many of them can simultanously connect to an AP.

kelson42, 2021-07-25:
One other thing which can be done, is buying/configuring 32 ESP8266 chipsets in Micro Python. Let them automatically connect to a predefined wifi network and see how many of them achieve it. We could probably get the hardware for ~300 USD. If @rgaudin would be motivated to play with that, I think that might be a viable approach.

kelson42, 2021-07-25:
An other approach would be to buy a 32 ports USB hub https://www.amazon.com/Eyeboot-Port-240W-Powered-Charging/dp/B07TLBWRZ6, plug 31 wifi dongles, and run a script on a linux computer (or instrumented RPI).

kelson42, 2021-07-25:
TL-WN725 seems to be a pretty good dongle for our use case https://www.amazon.com/gp/product/B008IFXQFU/ref=as_li_qf_asin_il_tl?ie=UTF8&tag=linuxhint-20&creative=9325&linkCode=as2&creativeASIN=B008IFXQFU&linkId=165bb901a0118279f383b80ee2702776#customerReviews

kelson42, 2021-09-24:
/move offspot/testbench

closed 2021-09-24 by move[bot]