offspot/kiwix-hotspot #545: Does the hotspot support ethernet connection?

Popolechien, 2020-11-12:
My understanding of #215 and #223 is that a hotspot plugged via ethernet to an existing connection will be able to sort things and pick content online of offline depending on whether it has the content locally or not.
- Is this assumption correct?
- Is this documented somewhere?

rgaudin, 2020-11-12:
I don't think I understand the question properly.

- hotspot content is set at creation time.
- homepage is built at creation time with list of content selected.
- so there's no such thing as presenting content that is not on the card.

Ethernet connection, if working and connected to Internet shares that internet connection with the hotspot user. Does that answers your question?

closed 2020-11-12 by rgaudin

Popolechien, 2020-11-12:
Ok that partially answers my question, but let me describe my use case:
- Hotspot is connected via Ethernet to Internet.
- User connects to "Hotspot" SSID
- User goes to https://www.google.com -> works
- User goes to http://goto.kiwix.hotspot -> works as well

Would this work or is there some intermediate step?
Follow-up question: can we setup the hotspot so that it connects to an existing wifi network and behaves the same as with a ethernet cable?

rgaudin, 2020-11-12:
No other step than the captive portal.

We have no "setup" in hotspot so, no. eth is for internet, wifi is for hotspot.

We could imagine a feature that let you reverse but that would mean config on running Pi instead of in installer.