offspot/kiwix-hotspot #546: Display tutorial to invite siwtching to HTTPs for SW based ZIM

kelson42, 2020-11-13:


rgaudin, 2021-05-07:
Please elaborate

kelson42, 2021-05-07:
If the Hotspot welcome page is in HTTP and the user clicks on a SW ZIM, then a warning page should be displayed we following properties:
* Inform the user that he is going to move to HTTPS
* He is going to get a warning from his browser (one sentence to explain a bit why).
* It will look like this, display one/two pictures per browser
* At the end, "I have understood and I'm ready to proceed".

Popolechien, 2021-05-07:
Also I would replace the "I have understood and I'm ready to proceed" with a simple "ok" button. People don't read it, it's not like they're given a choice, and after 2-3 occurences that much wordiness might feel like a drag.

rgaudin, 2021-05-07:
@Popolechien it's not possible to do it transparently, otherwise there would be no ticket.

It's not a browser warning, it's a browser danger UI that requires manual action that consists of several clics. Not preparing users would mean they'd just go back and thus no SW ZIM use.

closed 2021-05-14 by rgaudin