offspot/kiwix-hotspot #548: Installer crashes when a wifi password is configured

Popolechien, 2020-11-14:
See e.g., job [L361R5FAF](https://cardshop.hotspot.kiwix.org/orders/L361R5FAF): wifi password is set to `edtech ` (without any special character and the installer fails.
Tried the exact same config without password (ie open wifi) and it worked fine.

Popolechien, 2020-11-14:
Password had six characters (too short).

kelson42, 2020-11-14:
It should not crash, keeping ticket open.

rgaudin, 2020-11-14:
duplicate

kelson42, 2020-11-14:
This should be fixed in cardshop as well (securing password size).

rgaudin, 2020-11-14:
it doesn't crash

kelson42, 2020-11-14:
@rgaudin @Popolechien A bit of clarity in your comments would be welcome. I don't understand:
* what is the exact symtom of the problem
* where is the duplicate.

I have found a duplicate at cardshop level https://github.com/kiwix/cardshop/issues/83 but nothing at kiwix-hotspot installer level.

rgaudin, 2020-11-14:
there is no crash of the hotspot so nothing to fix here.

kelson42, 2020-11-14:
Ok, lets try to make it clear: what happens exactly when we put a 6 character long password? It seems that the wifi won't be usable, so we should build a security to avoid that.

Popolechien, 2020-11-14:
I think it loops back to what was said in https://github.com/kiwix/cardshop/issues/210 : we should prevent users from doing things we do not want them to.

To answer your question, if the password is incorrect (too short, or with special characters), the installer (not the hotspot) fails.

kelson42, 2020-11-14:
Exactly, we should prevent this in Kiwix-hotspot. The other ticket is at the cardshop.

closed 2020-11-14 by kelson42