offspot/kiwix-hotspot #552: Larger .zim files not searchable

Bahusson, 2020-11-23:
On 256GB image. Around 5GB memory left unused.

There is a problem with english Wikipedia search.

Khan appears to work and Polish wiki even gives you suggestions as you search, but once you try to make a direct search in english wiki it throws a timeout and all I can do is restart device. 
I'm still able to search english wiki by clicking on links however, so it's not entirely broken. It's just something with the search.

rgaudin, 2020-11-23:
Thank you, there no link between this and the free space available on the card.
This issue is a kiwix-serve issue (hosted in kiwix-tools or kiwix-lib repo) and I believe it's a known issue. @kelson42 will move it accordingly

kelson42, 2020-11-23:
@Bahusson Does the whole suggestion syste fails or only the fulltext search?

Bahusson, 2020-11-23:
@kelson42 For full english wikipedia .zim the search suggestions do not appear at all regardless of how long I wait for suggestion to appear. In full polish wiki the suggestion appears instantly.

kelson42, 2020-11-24:
@rgaudin @Popolechien Can one of you test with aPI zero?

rgaudin, 2020-11-24:
I can't download such large files and I don't have SD card this large neither. Good to know the Pi0 supports those though.

Popolechien, 2020-11-24:
I've launched a creation, will report back when tested.

kelson42, 2020-11-26:
@Popolechien Have been able to reproduce the problem?

Popolechien, 2020-11-26:
Am still waiting for the card to be generated : got notification that it's been uploaded but I suspect it's now waiting for room on the writer or some similar recurring issue.

rgaudin, 2020-11-26:
Checked and there was an unrelated issue: the worker couldn't resolve `api.cardshop.hotspot.org`… but the host could. Maybe there had been a change at sloppy that was cached somehow. restarted the worker.

kelson42, 2020-12-01:
@Popolechien news?

Popolechien, 2020-12-18:
Finally could give it a look, and basically come up with the same issue as described. 
- Generated card including wikipedia_en_all_maxi (latest version, so probably 2020-11 which should have zstd)
- Ran it on Raspberry Pi W
- Searched for a random topic
- Got 504 Gateway timeout after a minute.
- Pi crashes and needs to be restarted.

rgaudin, 2020-12-18:
Thanks for the report. Should we move that to kiwix-tools/library?

kelson42, 2020-12-18:
/move kiwix/kiwix-tools

closed 2020-12-18 by move[bot]