offspot/kiwix-hotspot #553: Library does not refresh

Popolechien, 2020-11-25:
A couple of files were generated on the zimfarm about 48hrs ago and are still not available for image building (tested on cardshop, but I suspect the library issue is an installer issue)

kelson42, 2020-11-26:
This is a consequence of have library.kiiwx.org not refreshed. I have fixed the problem and from next night, it should be back to the normal.

Popolechien, 2020-12-01:
Reopening the issue as the Greek and Arabic Core PHZH moocs are not to be found (neither in their home language nor in English). They are available on library.kiwix.org though

kelson42, 2020-12-01:
@Popolechien What is the problem exaclty? What do you do for searching them? With which Kiwix? What is the ZIM file name you want to fin?

Popolechien, 2020-12-01:
@kelson42 I'm on the cardshop and these two zim files are not listed in the list of zim files (though they were produced 8 days ago) so I can't include them in a config.
Am looking for these two: phzh_core-arabic-one-test-zim_ar_2020-11 (Core Arabic 1) and phzh_core-greek-one_el_2020-11 (Core Greek 1)

kelson42, 2020-12-01:
They are both in the Kiwix-Hotspot library, see https://mirror.download.kiwix.org/library/ideascube.yml

kelson42, 2020-12-01:
@rgaudin I confirm that if I search content based on language `eng` (because of https://github.com/openzim/openedx/issues/165), and search in page "core", I find none of the title given above.

rgaudin, 2020-12-01:
[image]

rgaudin, 2020-12-01:
Thanks, I've looked at it and can confirm there's an issue in the cardshop. The YAML catalog is stored in a cache that is set to expire after 24h. The cache has the new value and I believe it's being refreshed properly.
I think the problem lies in the temp variables holding filtered version of the data and the process lifespan.
restarted the container manually so you can see new values until this is fixed.