offspot/kiwix-hotspot #556: Cannot Boot on Raspberry Pi 4B

michaelcox, 2021-01-19:
I signed up through the [Cardshop Access](https://www.kiwix.org/en/cardshop-access/) and created a custom image that I downloaded and burned to a 256Gb SD card.

That card works just fine when I put it into my Raspberry Pi Zero W, but will not boot from my Raspberry Pi 4B 2Gb.

Here is a photo of the boot screen:
[image]

The Pi 4B does work fine booting from a separate SD card with Raspberry Pi OS on it, and I've updated everything via Pi OS, so it doesn't immediately look like a hardware issue. I contacted via Support and they recommended I post here.

Since I didn't create the underlying image (just downloaded from the Hotspot Cardshop), I'm unsure how to troubleshoot. Thanks for your help.

rgaudin, 2021-01-20:
Thank you for the report. It seems the firmware that's included in the raspbian version we use is not compatible with your HW.

It does work on bare Pi4B.
It doesn't seem there has been a new HW revision so I think maybe newest version of the OS/PiOS updated the bootloader (I see a very recent September 2020 in your picture) and that new bootloader expects newer firmware.

The one we bundle dates back July 2019.

```
-rwxr-xr-x 1 root root 2759172 Jul  9  2019 start4.elf

```

[This page](https://www.raspberrypi.org/documentation/hardware/raspberrypi/booteeprom.md) clearly describes this.

I think the action plan would be:

- Mark hotspot/cardshop as not compatible with Pi4 anymore
- fix kiwix-hotspot to bundle newer version of the firmware
- find a strategy to be informed of new bootloader/firmare versions and update kiwix-hotspot every time
- document that using a different sd-card (PiOS at least)  on an already setup hotspot might lead to it not being able to start anymore (PiOS checks for firmware/bootloader updates on start and applies on next reboot).

Popolechien, 2021-01-20:
@rgaudin I've updated the cardshop page to indicate that Pi4 built late 2020 are not compatible (mine was bought in June and works fine).

kelson42, 2021-01-20:
> find a strategy to be informed of new bootloader/firmare versions and update kiwix-hotspot every time

Watch https://github.com/raspberrypi/rpi-eeprom ?

michaelcox, 2021-01-20:
Thanks everyone. Let me know if you need me to help you test anything on my board. Or if I can help in any other way.

Popolechien, 2021-01-20:
@michaelcox Thanks! We'll most certainly come back to you once this is updated and would appreciate if you can test on your RPi.

Di9italZero, 2021-03-15:
Is there any way to get Kiwix working on the RPI 4?

rgaudin, 2021-03-15:
If you're only interested in Kiwix-serve, you can use the latest raspbian with [kiwix-tools](https://download.kiwix.org/release/kiwix-tools/) ; those are static binaries.

Di9italZero, 2021-03-15:
Thanks. So I can serve kiwix from Raspbian using kiwix-tools. Will
kiwix-tools create the necessary hotspot as well. Thanks for the reply.

rgaudin, 2021-03-16:
No kiwix-serve will just serve zim files.

Di9italZero, 2021-03-16:
What a joke. Support your projects. Don't advertise things that don't work. There are thousands of people who wanted to build this. They studied the online tutorials and purchased hardware all to find out kiwix can't fix their image. Gtfoh. This is bull.

rgaudin, 2021-04-15:
We've now added support for all Pi4 and Pi400 by creating our master off the latest raspiOS. Will be available soon on 2.4.0 release.

closed 2021-04-15 by rgaudin