offspot/kiwix-hotspot #558: Add "external links" off switch

Popolechien, 2021-01-26:
From a user: _"I noticed that clicking links that go out to external pages with an "https" in the URL delivers an error message that's a bit disconcerting to non-technical users."_
[image]

Since most use cases relate to people being offline, there is no point in letting them click on links to nowhere that will worry them more than anything. Could it be possible to have a switch feature before building the image that inactivates these external links by default (but can be reactivated if need be)?

rgaudin, 2021-01-26:
For kiwix-serve, we have a feature (not enabled on hotspot) that catches external links and displays a warning. We could enable that on the hotspot and let them know that if they get the warning, it's because the hotspot is not connected to Internet and this the external link is useless.

A second iteration could check the Internet connection status and behave differently (either different message or just redirect to the destination).

kelson42, 2021-01-26:
@rgaudin Yes this is probably a bettre approach to configure this per default as the use case where a Kiwix-Hotspot is online is a rare one.

HassanRahim26, 2021-03-10:
Hi, I am Hassan. I would like to work on this issue, can you suggest ways to get started?

SRDdev, 2021-06-05:
Can I work on this issue! 
@Popolechien

kelson42, 2021-06-05:
@SRDdev Yes, would you be able first to explain here how your solution woukd look like?

SRDdev, 2021-06-07:
Sorry I thought about it and found some technical issues in my solution, you can pass it on @kelson42 
Thank you so much @kelson42 !!

Popolechien, 2021-06-07:
@SRDdev if you found the issues, then you are halfway to solving them ;-)

kelson42, 2024-06-04:
Fixed in Kiwix-Hotspot 3.0